uzh-rpg/flightmare #7: Problem completing Basic usage with ROS

AlessandroSaviolo, 2020-09-03:
Hi!

I've done all the installation steps as suggested by the guide "Basic usage with ROS", however I couldn't launch the file `rotors_gazebo.launch` (the launch was successful but the Flightmare gui wasn't able to connect to anything).. I've tried to launch `rotors_gazebo_test.launch` and it worked.

By commenting piece by piece of the `rotors_gazebo.launch` launch file, I've figured out that the problem was about the Gazebo nodes and the line https://github.com/uzh-rpg/flightmare/blob/master/flightros/launch/rotors_gazebo.launch#L27.

Is this line of code necessary for any purpose?

AlessandroSaviolo, 2020-09-04:
Update: By creating a new workspace with only this simulator, the problem is solved. Probably some 

In order to use the PAMPC controller, the autopilot node in the launch file (https://github.com/uzh-rpg/flightmare/blob/master/flightros/launch/rotors_gazebo.launch#L72) must be skipped if the flag use_mpc is true (i.e., `<group unless="$(arg use_mpc)"></group>`)

yun-long, 2020-09-04:
> Update: By creating a new workspace with only this simulator, the problem is solved. Probably some conflict in the dependencies..
> 
> In order to use the PAMPC controller, the autopilot node in the launch file (https://github.com/uzh-rpg/flightmare/blob/master/flightros/launch/rotors_gazebo.launch#L72) must be skipped if the flag use_mpc is true (i.e., `<group unless="$(arg use_mpc)"></group>`)

Great. Happy that u addressed the issue. 
For using the mode based controller, you can visit the rpg_quadrotor_control or rpg_mpc for more details.

closed 2020-09-04 by yun-long