uzh-rpg/flightmare #11: Failed to build flightgym

xjp99v5, 2020-09-05:
Command errored out with exit status 1: /home/jiaping001/anaconda3/envs/Flightmare/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-vlc9823y/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-vlc9823y/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-r94opyvc/install-record.txt --single-version-externally-managed --compile --install-headers /home/jiaping001/anaconda3/envs/Flightmare/include/python3.6m/flightgym Check the logs for full command output.

xjp99v5, 2020-09-06:
I am using ubuntu20, conda4.8.4 and opencv4. I cannot build flightgym using these environments.

xjp99v5, 2020-09-07:
I have solved this problems  by looking at the opencv directories

yun-long, 2020-09-07:
> I have solved this problems by looking at the opencv directories

That's great. I never tested it on Ubuntu20, I assume it should also work. 
Maybe you can provide more details about how you solved the problem. might be useful for other people who is using Ubuntu20.
Thanks.

xjp99v5, 2020-09-07:
When I was using ubuntu20, conda4.8.4 and opencv4 environment, I just added "include_directories(${OpenCV_INCLUDE_DIRS})" after "find_package(OpenCV REQUIRED)" in /flightmare/flightlib/CmakeLists.

yun-long, 2020-09-07:
closing the issue. If this problem appears again, I will look into it.

closed 2020-09-07 by yun-long