uzh-rpg/flightmare #12: Cannot connect the Unity when run python3 run_drone_control.py --train 0 --render 1

xjp99v5, 2020-09-07:
Loading a model without an environment, this model cannot be trained until it has a valid environment.

yun-long, 2020-09-07:
Hi, Could you provide more details? 
It is difficult for me to see what exactly is the problem. 
However, you shouldn't enable `render` for training. It makes data sampling extremely slow. 
Doesn't the training work for you if you put -render 0?

xjp99v5, 2020-09-07:
Yeah. When input "python3 run_drone_control.py --train 1" in terminal, it could work. However, when I input "python3 run_drone_control.py --train 0 --render 1", the terminal shows "/home/jiaping/anaconda3/envs/flightmare/lib/python3.6/site-packages/cloudpickle/cloudpickle.py:704: UserWarning: A pickle file created using an old (<=1.4.1) version of cloudpicke is currently being loaded. This is not supported by cloudpickle and will break in cloudpickle 1.7
  "will break in cloudpickle 1.7", category=UserWarning
Loading a model without an environment, this model cannot be trained until it has a valid environment.
" and pops up a figure generated by Matplotlib. The Unity engine has no action. I was wondering it is because of the version of anaconda. Which version of anaconda are you using?

yun-long, 2020-09-07:
I see. `Loading a model without an environment, this model cannot be trained until it has a valid environment.` is actually a warning message from `_stable-baselines._` 
It has nothing to do with Unity or Flightmare.

By running `python3 run_drone_control.py --train 0 --render 1` you are simply running a pre-trained model and enabled the Unity. It should try to connect the Unity App and then load the environment. 
It was my fault. Now the problem is fixed. 

Try to pull the master branch, and do the pip install for both the `flightlib` and the `flightrl` again. It should work.

xjp99v5, 2020-09-07:
Okay. Thank you a lot.

closed 2020-09-07 by xjp99v5