uzh-rpg/flightmare #15: Abandoned: 15 packages were abandoned.

sotomotocross, 2020-09-08:


yun-long, 2020-09-08:
you are missing the `catkin_pkg` in your python. Try to install it via "(sudo) pip install catkin_pkg" or "pip3 install catkin_pkg". 

Or google the error message, **Import error, no module named "catkin_pkg"**. I believe you can find the answer.

sotomotocross, 2020-09-09:
I have run all the commands you have told me.
Also I tried to run: sudo apt-get install python-catkin-pkg
In all cases the catkin_pkg is installed in the system.
Also googling the message didn't gave me any solutions.
Do you have any more propositions?
Thank you very much.

yun-long, 2020-09-11:
Hmmm, it looks like a catkin/python/ROS problem. 

Did you build other ROS projects on your machine before?

peasant98, 2020-09-12:
Hi @sotomotocross, do you get a new error after installing `catkin_pkg`? Or does `from catkin_pkg.package import parse_package` still cause an error?

yun-long, 2020-09-18:
closing the issue. it seems like a ROS installation problem. feel free to reopen it if the issue still exists.

closed 2020-09-18 by yun-long

sotomotocross, 2020-09-23:
Excuse me for the lack of feedback. I was with another project.
The error still exists. I tried to install the flightmare on a new Ubuntu 18.04.05 LTS system and i have the problem of the two attached screenshots.
[image]
[image]
I have installed the below:

1. sudo apt install python-catkin-pkg python-catkin-tools
2. pip install catkin-tools
3. pip install catkin-tools-python
4. sudo -H pip install -U catkin_pkg

Still the error doesn't stop appearing.
I have run catkin build before on this system and everything run ok. Especially earlier today i built https://github.com/uzh-rpg/rpg_esim that uses similar steps.

I don't know what else to do and i really need any help available.
Thank you in advance for the cooperation, time and help.
Looking forward to here from someone.

vivekrathi14, 2020-11-05:
Initialize your build for python 2.7. It seems your catkin tools point to python 3.7.
Do this in your workspace directory (~/catkin_ws)
`catkin config --init --mkdirs --extend /opt/ros/$ROS_DISTRO --merge-devel --cmake-args -DCMAKE_BUILD_TYPE=Release -DPYTHON_VERSION=2.7`