uzh-rpg/flightmare #16: Airsim instead of Unity

hai-h-nguyen, 2020-09-08:
How about using Airsim instead of Unity?

peasant98, 2020-09-08:
Do you mean Unreal Engine @hai-h-nguyen? This repo is more of an alternative to using drones in Airsim.

hai-h-nguyen, 2020-09-09:
Yeah, I mean using Unreal Engine. We are kind of having the whole scene in Airsim already but this repo comes out and it is promising. We don't want to rebuild the whole thing in Unity so I wonder if there is a way to utilize what we have in Airsim.

peasant98, 2020-09-12:
Has the work you've done been tightly coupled with the Airsim environment (Unreal Engine, other Airsim related stuff)?

peasant98, 2020-09-18:
@hai-h-nguyen

yun-long, 2020-09-18:
Hi, It depends on your application. For machine learning applications, such as collect images with segmentation for training neural networks, AirSim might be better. In the cases of agile quadrotor fly, reinforcement learning, path planning, optimal control, you can consider using Flightmare. Check our paper for more details.    https://arxiv.org/abs/2009.00563

hai-h-nguyen, 2020-09-18:
@peasant98 We decided to go with Airsim since we want to control directly from pixels. 
@yun-long  is the parallelism in this repo means running a scene with many drones? It is quite different from the parallelism in Gym.

yun-long, 2020-09-18:
@hai-h-nguyen yes.

closed 2020-09-18 by yun-long