uzh-rpg/flightmare #19: no module flightgym

yananliusdu, 2020-09-11:
Hi Yunlong,

thanks for your excellent work with the Flightmare simulator which would be a good contribution to the realm of robotics.

btw, when I setting up with python after pip install,  and try to run an example by 'python3 run_drone_control.py --train 1', it always reports there is no module 'flightgym'. 

do you know the possible reason for that?  thanks.

yun-long, 2020-09-11:
did you install `flightlib` by following this https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip?

yananliusdu, 2020-09-12:
Hi Yunlong,  

thanks for your timely reply and I managed to solve this issue with your reminding. 

however, at the last step to run your simulator with  '[Basic Usage with Python|Basic-Usage-with-Python]',  this bit specifically "python3 run_drone_control.py --train 0 --render 1", 

it reports  '[UnityBridge]  Unity Connection time out! Make sure that Unity Standalone or Unity Editor is running the Flightmare.'. I think I had put the extracted unity folder into the right place (in flightrender folder and I also saw the connection UI after double-clicking RPG_Flightmare.x84-64 )

[image]

do you know the possible reason resulting in this?

thanks.

yun-long, 2020-09-14:
Hi,

For this, make sure that you open the Unity standalone first (double click RPG_Flightmare.x86-64), and then, run "python3 run_drone_control.py --train 0 --render 1"

yananliusdu, 2020-09-14:
Hi Yunlong,

I repeated this several times. it seems like the UI pops out and then disappear within several seconds after double-clicking.  and after running "python3 run_drone_control.py --train 0 --render 1", it ends up with "Unity connection time out". it that because I used a virtual machine or something?

thanks.

yun-long, 2020-09-18:
Hi, to be honest, I don't know. It might be the reason. But this is for sure Unity Issue.  Do you have an Ubuntu machine and try it out on Ubuntu 18.04, since we have heavily tested Flightmare on that.

sansha, 2020-12-11:
Hi,

I can confirm this. I couldn't get it to work in my Ubuntu 18.04 machine. the only output when executing it in the command line is
```
Set current directory to /home/ros-industrial/flightmare_catkin_ws/src/flightmare/flightrender
Found path: /home/ros-industrial/flightmare_catkin_ws/src/flightmare/flightrender/RPG_Flightmare.x86_64
```

on my native Ubuntu 20.04 it works (it starts the main screen of the simulator).

maybe you could add this as a short note to the quick start documentation, it cost me a whole day to figure out the VM was the problem. Thanks!

If I find a workaround, I'll also post it here.