uzh-rpg/flightmare #20: Multiple Quadrotors

ghokulji, 2020-09-14:
I am using flightmare with ROS and gazebo. What changes are to be made to launch multiple quadrotors in the flightmare environment and extract pointcloud and camera data?

Also,I have a few questions regarding the flightmare simulator:

1) How to added Lidar and camera capabilities to the rpg quadrotor and extract the data in the flightmare environment? Also how to add obstacles to the flightmare scene?
2) What is the purpose of using  flightmare with ROS and gazebo, why can't we directly use ROS with flightmare unity?
3) Is it possible to launch mobile wheeled robots( eg. Clearpath Husky robots) in the flightmare simulator?

Thanks!

yun-long, 2020-09-18:
Hi,

To answer your question:

0.1  For launch multiple quadrotors in the Flightmare, we only have examples in Reinforcement learning vectorized environment. https://github.com/uzh-rpg/flightmare/blob/master/flightlib/src/envs/vec_env.cpp#L94-L97  . it is not straightforward though.  If you don't need parallel processing, you can simply create multiple quadrotors (https://github.com/uzh-rpg/flightmare/blob/master/flightros/include/flightros/flight_pilot.hpp#L55).  and add it to Unity (https://github.com/uzh-rpg/flightmare/blob/master/flightros/src/flight_pilot.cpp#L84)

0.2 Please see flightlin/tests/bridges/unity_bridge.cpp for examples on how to extract point cloud and camera data. 

1. Lidar is not available at the moment. See  https://github.com/uzh-rpg/flightmare/blob/master/flightros/src/flight_pilot.cpp for examples. 
2. Gazebo provides more flexible and high-performance physics engines. In the case you don't need Gazebo, you can also use ROS with Flightmare, without enabling the Gazebo or RotorS simulation. 
3. It is possible. However not straightforward.

ghokulji, 2020-09-18:
Hey yun,

Thanks a lot for the detailed response. I will try out the various methods you have mentioned.

closed 2020-09-18 by yun-long

anoopsonar30, 2021-03-22:
I added multiple quadrotors but the camera only follows one of them. How would I go about keeping the camera static/fixed at some position in the environment? Or have 1 camera for each drone?