uzh-rpg/flightmare #22: error: ‘CV_RGB2BGR’ was not declared in this scope

ghokulji, 2020-09-16:
[image]

I tried to build the new commit, the above error is showing. I included the '#include <opencv2/imgproc/types_c.h> in the unity_bridge.hpp' and it builds without any error. Is it the right way to fix the error?

yun-long, 2020-09-16:
Hi @ghokulji , thanks for the feedback. 

I cannot reproduce the error. Which opencv version are you using?

ghokulji, 2020-09-16:
Hey yun,
I am using opencv version 3.2.

yun-long, 2020-09-16:
Hmmm, I am using the same OpenCV version. If it works, then I would say you are doing it right.

closed 2020-09-16 by yun-long

Josias-C, 2020-09-22:
Getting the same error with opencv version 4.2.0. The mentioned fix worked for me too.