uzh-rpg/flightmare #23: Ubuntu 20.04 and OpenCV 4.2 compilation issues

lbfd, 2020-09-18:
To install the flightmare simulator I precisely followed the steps mentioned on https://github.com/uzh-rpg/flightmare/wiki/Install-with-ROS on a fresh Ubuntu 20.04 install. However, I ran into a four problems which I'll detail below along with my fixes for them.

**First Issue: ROS Dependencies**
In Ubuntu 20.04 there is no `python-vcstool` to install. It needs to be `python3-vcstool

**Second Issue: OpenCV 4.2**
The current version of OpenCV is 4.2. In the `flightmare/flightlib/src/bridges/unity_bridge.cpp` file, line 237 the call to `cvtColor` gives a compile-time error because `CV_RGB2BGR` is not defined in OpenCV 4.2. According to https://docs.opencv.org/4.2.0/d8/d01/group__imgproc__color__conversions.html it needs to be `cv::COLOR_RGB2BGR`. 

**Third Issue: Python Syntax Error**
In line 53 of the file `rotors_simulator/rotors_evaluation/src/rosbag_tools/analyze_bag.py` the python syntax `except Exception, e` is not recognized when building the package. It works if the line is changed to `except Exception as e`

**Fourth Issue: OpenMP**
`Errors     << flightros:make /home/bfd/catkin_ws/logs/flightros/build.make.000.log                                                                                                                        
/usr/bin/ld: /home/bfd/catkin_ws/install/lib/libflightlib.so: undefined reference to 'omp_set_num_threads'
collect2: error: ld returned 1 exit status`
`make[2]: *** [CMakeFiles/flight_pilot_node.dir/build.make:181: /home/bfd/catkin_ws/devel/.private/flightros/lib/flightros/flight_pilot_node] Error 1`
`make[1]: *** [CMakeFiles/Makefile2:241: CMakeFiles/flight_pilot_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2`

It seems to like the `-fopenmp` flag is missing in the `flightmare/flightros/CMakeLists.txt`. Adding a `-fopenmp` flag to line 29 resolved the issue and catkin build runs without any errors.

ghokulji, 2020-09-22:
For the second issue mentioned about OpenCV, I included the header #include <opencv2/imgproc/types_c.h> in the unity_bridge.hpp  file. It built without any errors.

andrearlotta, 2022-06-22:
_It seems to like the -fopenmp flag is missing in the flightmare/flightros/CMakeLists.txt. Adding a -fopenmp flag to line 29 resolved the issue and catkin build runs without any errors._

How I can add this flag?

`SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fopenmp=TRUE")` ?

Thanks. Grazie