uzh-rpg/flightmare #25: How to fix the camera on the quadrotor？

RicardoZheng, 2020-09-22:
Hello,

Yunlong, thank you for sharing this simulator. 
I have seen flight_pilot.cpp where an RGB camera is added, and I tried to publish the RGB_image like the example in the test folders. But the camera can not move with the quadrotor, it seems like the pose of the camera is stationary but not fixed on the quadrotor. I would like to know how to fix the camera on the quadrotor？
I run Flightmare according to the Basic Usage with ROS in the wiki.

Thank you

goromal, 2020-09-22:
Throwing out there that I had this same issue until I downloaded the latest renderer version from the releases page.

yun-long, 2020-09-23:
Hi, 

Please download the latest render version from the releases page.

RicardoZheng, 2020-09-23:
Yes, the latest render version can fix the issue. Thank you all for your help.