uzh-rpg/flightmare #26: Problem completing the Basic Usage with ROS

sotomotocross, 2020-09-23:
I have successfully made catkin build my workspace (only one workspace for the flightmare simulator) and i try to complete the Basic Usage with ROS.
I downloaded RPG_Flightmare, i extracted it into the /path/to/flightmare/flightrender and run roslaunch flightros rotors_gazebo.launch as indicated.
Therefore, i have and Error in REST request and also the Unity Connection time out! 
[image]
Is there a proper way of extracting the contents of RPG_Flightmare? Should i change something or the extracted just has to be inside the flightrender directory?
Is there anothe problem?
Also if i run: roslaunch flightros rotors_gazebo_test.launch
then i have the error below:
[image]
Looking forward for any help.
Thank you very much.

sotomotocross, 2020-09-23:
Quick correction i changed the name of RPG_Flightmare.x86_64
The roslaunch flightros rotors_gazebo_test.launch runs ok.
Although if i run roslaunch flightros rotors_gazebo.launch all the errors indicated at the beginning remain.

yun-long, 2020-09-23:
can you first start the Unity Standalone by double-clicking the RPG_Flightmare.x86_64, and then, roslaunch flightros rotors_gazebo.launch?

sotomotocross, 2020-09-23:
Yes i double-clicked on the RPG_Flightmare.x86_64 and then roslaunch flightros rotors_gazebo.launch and everything started ok.
I hadn't understood that i have to run it like this before beginning the ROS.
Thank you very much.

yun-long, 2020-09-23:
It is implemented in the roslaunch file, https://github.com/uzh-rpg/flightmare/blob/master/flightros/launch/rotors_gazebo.launch#L68-L69

You can change the default value for `use_unity_editor`  to false (Which means you don't run unity editor but run the unity standalone called RPG_Flightmare.x86_64). The ros will launch RPG_Flightmare.x86_64 automatically.

sotomotocross, 2020-09-24:
I changed the value of use_unity_editor to false and the  roslaunch flightros rotors_gazebo.launch worked very good.
Thank you very much for the help.

Reka-Berci-Hajnovics, 2020-10-08:
I have run into the same issue. @yun-long  Could you please consider adding a Troubleshooting section in the _Basic usage with ROS_ page?
I think it would be a helpful tip for other users as well.

vivekrathi14, 2020-11-05:
Hi,
I am having similar issue;

System Specifications -
VMWare Ubuntu 18.04, ROS Melodic

I followed all instructions from wiki and was able to install and build the packages in the workspace.
Further I downloaded the RPG_Flightmare and extracted it into `<workspace>/src/flightmare/filghtrender`
I tried running` roslaunch rotors_gazebo_test.launch`

It gives me this **error**:
`[rpg_flightmare_render-3] process has died`

I tried to run the executable but I get this error:
[image]

However, if I use sudo ./RPG_Flightmare.x86_64 it works.

Can someone help me here to get ROS example working?