uzh-rpg/flightmare #27: Where is documentation?

NicolasMelot, 2020-09-23:
Hi,

The flightmare paper makes all sorts of very nice feature claims, most notably for me an API that allows access to ground truth point cloud and a collision-free motion planner. However I cannot find any documentation on how to use these two, or anything else beyond the basic usage with ROS. Can you point me to more extended information on how to get started there?

yun-long, 2020-09-23:
Hi

The point cloud is currently available. There are two options:

1. use C++ client to extract the point could. see https://github.com/uzh-rpg/flightmare/blob/master/flightlib/tests/bridges/unity_bridge.cpp#L27-L42 for an example. 

2. Use the UI to extract the point cloud. Please download the latest standalone from the **Release.**  extract it to the flightrender folder or anywhere you want. Double click the **RPG_Flightmare.x86_64**. Then you can extract the point cloud directly using the UI. 

Yes, it is possible to do collision-free motion planner with the point cloud, however, the example code is currently not available. 
We will publish the code at some point, but currently we are focusing on more important stuff. 

You can check the https://ompl.kavrakilab.org/tutorials.html for tutorials and write your own planner.

NicolasMelot, 2020-09-23:
Is there any doc on how to use RPG_Flightmare.x86_64 to extract the cloud point? I can run it and have ros/gazebo to insert and run a drone in it, but nothing else.
I suppose what the OMPL code can give me is an ordered list of 3D coordinates (+orentiation) to move so that the drone can achieve a collection-free trajectory between arbitrary points A and B. However I cannot find the API that allows me to move the drone to a given coordinate. Do you have more doc or code example pointers for me?

yun-long, 2020-09-24:
Hi, 

In order to use the Point-Cloud extraction UI, you don't need to run ROS. Simply download the newest RPG_Flightmare.x86_64 (version 0.0.3). After starting the Unity Standalone, you will see the point cloud button. 

I understand that this is not straightforward. There is no simple API for your task, but you can 
   1) find the collision-free path using OMPL (path planning)
   2) fit the path with polynomial trajectory (trajectory generation)
   3) Use rpg_quadrotor_control to control the quad to follow this trajectory (optimal control)

NicolasMelot, 2020-09-24:
Hi,

This might keep me busy for a while. Thank you for the pointers