uzh-rpg/flightmare #28: Running flightmare with px4 sitl

sandeshthapa, 2020-09-24:
Hi I'm trying to run flightmare with ROS and gazebo, I was wondering if I can run flightmare with px4 sitl, like similar to mavros nodes. If there is any way please guide me. Thanks.

sandeshthapa, 2020-09-24:
Hi, 

I'm running flightmare with ros and gazebo. I'm wondering if there is a way to run px4 sitl with flightmare.  If there is as way please, guide me. Thanks.

yun-long, 2020-10-05:
HI,

Sorry, I don't know the answer to your question, since I personally don't use PX4 and am not familiar with PX4.

Reka-Berci-Hajnovics, 2020-10-08:
Hi,
At our company, we are using ROS, Gazebo, and PX4 for all our drone-related simulations. The flightmare looks really amazing! It is a pity that it is not compatible with PX4. Are you considering connecting it with PX4? 
If not could you help with where to start to make it compatible?

closed 2024-12-21 by sandeshthapa