uzh-rpg/flightmare #35: catkin build fails

Reka-Berci-Hajnovics, 2020-10-08:
I was following the instructions on the wiki to install fligthmare with ROS, but when I try to build it catkin build fails.
I get the following error message:
[image]

I've tried to edit the PYTHONPATH, but then it messes up my pip installation.

After trying couple of different solutions I have ended up removing my build and devel folder and I ran `catkin_make`.
`catkin_make` was able to build my packages without any problems.

I think this tip might be handy for other users as well. Would you consider adding it to your documentation?
Thanks!

mrwrongwrongwrong, 2020-10-10:
The exactly same issue happened here, only 20 out of 33 packages succeeded.  Thank you much for bringing it up along with your solution.

mrwrongwrongwrong, 2020-10-10:
Dear Reka, 
     In addition, could you please supply the tutorial of " removing my build and devel folder"? I am a novice to kernel and trying to learn it. Thank you a lot for your help!
      Sincerely yours,

mrwrongwrongwrong, 2020-10-10:
> I was following the instructions on the wiki to install fligthmare with ROS, but when I try to build it catkin build fails.
> I get the following error message:
> [image]
>
> I've tried to edit the [...]

Dear Reka,
In addition, could you please supply the tutorial of " removing my build and devel folder"? I am a novice to kernel and trying to learn it. Thank you a lot for your help!
Sincerely yours,

mrwrongwrongwrong, 2020-10-10:
In case catkin_make does not work properly on your PC, the alternative solution is here: http://wiki.ros.org/catkin/commands/catkin_make

Reka-Berci-Hajnovics, 2020-10-12:
> Dear Reka,
> In addition, could you please supply the tutorial of " removing my build and devel folder"? I am a novice to kernel and trying to learn it. Thank you a lot for your help!
> Sincerely yours,

Hi,
Removing the build and devel folder is really easy. You can do it even without using the terminal. Just navigate to your catkin_ws, right-click the build and devel folder and choose _move to trash_.
From the command line you can do the same:

```
cd ~/catkin_ws
rm -r build
rm -r devel
catkin_make
```

vivekrathi14, 2020-11-05:
Initialize your build for python 2.7. It seems your catkin tools point to python 3.7.
Do this in your workspace directory (~/catkin_ws)
catkin config --init --mkdirs --extend /opt/ros/$ROS_DISTRO --merge-devel --cmake-args -DCMAKE_BUILD_TYPE=Release -DPYTHON_VERSION=2.7

MiquelEspinosa, 2021-01-09:
> I was following the instructions on the wiki to install fligthmare with ROS, but when I try to build it catkin build fails.
> I get the following error message:
> [image]
>
> I've tried to edit the [...]

Agree. I have tried with Ubuntu18.04, and `catkin build` command does not work properly. 
`catkin_make` works perfectly.

I think this should be mentioned in the documentation for future users.
Best regards,

MichaelChoum, 2022-12-22:
> Agree. I have tried with Ubuntu18.04, and `catkin build` command does not work properly. `catkin_make` works perfectly.
>
> I think this should be mentioned in the documentation for future users. Best [...]

Thank you very much! It works perfectly right now!