uzh-rpg/flightmare #38: Not able to Catkin build

PulkitRustagi, 2020-10-08:
virtualpullu@virtual-PulluBox:~/catkin_ws/src$ catkin build
-----------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/virtualpullu/catkin_ws
-----------------------------------------------------------------
Build Space:        [exists] /home/virtualpullu/catkin_ws/build
Devel Space:        [exists] /home/virtualpullu/catkin_ws/devel
Install Space:      [unused] /home/virtualpullu/catkin_ws/install
Log Space:         [missing] /home/virtualpullu/catkin_ws/logs
Source Space:       [exists] /home/virtualpullu/catkin_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-----------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-----------------------------------------------------------------
[build] Found '35' packages in 0.0 seconds.                                                         
[build] Updating package table.                                                                     
Starting  >>> catkin_simple                                                                         
Finished  <<< catkin_simple                                 [ 2.4 seconds ]                         
Starting  >>> mav_msgs                                                                              
____________________________________________________________________________________________________
Errors     << mav_msgs:cmake /home/virtualpullu/catkin_ws/logs/mav_msgs/build.cmake.000.log         
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "trajectory_msgs"
  with any of the following names:

    trajectory_msgsConfig.cmake
    trajectory_msgs-config.cmake

  Add the installation prefix of "trajectory_msgs" to CMAKE_PREFIX_PATH or
  set "trajectory_msgs_DIR" to a directory containing one of the above files.
  If "trajectory_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

cd /home/virtualpullu/catkin_ws/build/mav_msgs; catkin build --get-env mav_msgs | catkin env -si  /usr/bin/cmake /home/virtualpullu/catkin_ws/src/mav_comm/mav_msgs --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/virtualpullu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/virtualpullu/catkin_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
....................................................................................................
Failed     << mav_msgs:cmake                                [ Exited with code 1 ]                  
Failed    <<< mav_msgs                                      [ 2.3 seconds ]                         
Abandoned <<< mav_state_machine_msgs                        [ Unrelated job failed ]                
Abandoned <<< mav_system_msgs                               [ Unrelated job failed ]                
Abandoned <<< rotors_comm                                   [ Unrelated job failed ]                
Abandoned <<< rotors_description                            [ Unrelated job failed ]                
Abandoned <<< rotors_evaluation                             [ Unrelated job failed ]                
Abandoned <<< rqt_rotors                                    [ Unrelated job failed ]                
Abandoned <<< eigen_catkin                                  [ Unrelated job failed ]                
Abandoned <<< flightlib                                     [ Unrelated job failed ]                
Abandoned <<< flightrender                                  [ Unrelated job failed ]                
Abandoned <<< quad_launch_files                             [ Unrelated job failed ]                
Abandoned <<< quadrotor_msgs                                [ Unrelated job failed ]                
Abandoned <<< rpg_single_board_io                           [ Unrelated job failed ]                
Abandoned <<< mav_planning_msgs                             [ Unrelated job failed ]                
Abandoned <<< quadrotor_common                              [ Unrelated job failed ]                
Abandoned <<< polynomial_trajectories                       [ Unrelated job failed ]                
Abandoned <<< position_controller                           [ Unrelated job failed ]                
Abandoned <<< rotors_control                                [ Unrelated job failed ]                
Abandoned <<< rotors_gazebo_plugins                         [ Unrelated job failed ]                
Abandoned <<< rotors_gazebo                                 [ Unrelated job failed ]                
Abandoned <<< rotors_hil_interface                          [ Unrelated job failed ]                
Abandoned <<< rotors_joy_interface                          [ Unrelated job failed ]                
Abandoned <<< rpg_rotors_interface                          [ Unrelated job failed ]                
Abandoned <<< rqt_quad_gui                                  [ Unrelated job failed ]                
Abandoned <<< sbus_bridge                                   [ Unrelated job failed ]                
Abandoned <<< manual_flight_assistant                       [ Unrelated job failed ]                
Abandoned <<< state_predictor                               [ Unrelated job failed ]                
Abandoned <<< trajectory_generation_helper                  [ Unrelated job failed ]                
Abandoned <<< autopilot                                     [ Unrelated job failed ]                
Abandoned <<< flightros                                     [ Unrelated job failed ]                
Abandoned <<< rpg_quadrotor_integration_test                [ Unrelated job failed ]                
Abandoned <<< vbat_thrust_calibration                       [ Unrelated job failed ]                
[build] Summary: 1 of 33 packages succeeded.                                                        
[build]   Ignored:   2 packages were skipped or are blacklisted.                                    
[build]   Warnings:  None.                                                                          
[build]   Abandoned: 31 packages were abandoned.                                                    
[build]   Failed:    1 packages failed.                                                             
[build] Runtime: 5.0 seconds total.                                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.

peasant98, 2020-10-14:
This error most likely means that you're missing a ROS package; namely, the `trajectory_msgs` package. Link [here](http://wiki.ros.org/trajectory_msgs). Lmk if this helps!

PulkitRustagi, 2020-10-14:
Thanks! It did help in building a few more packagess but now its giving me this error. I am running this on a virtual machine(ubuntu 18.04), setup on windows, I don't think this would cause a problem but wanted to mention it none the less.

-------------------------------------------------------------------------------------------------------------

Errors     << flightlib:cmake /home/pullu/catkin_ws/logs/flightlib/build.cmake.001.log              
/usr/include/eigen3
CMake Warning (dev) at /usr/src/googletest/CMakeLists.txt:3 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/src/googletest/googlemock/CMakeLists.txt:40 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/src/googletest/googletest/CMakeLists.txt:47 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

ImportError: "from catkin_pkg.package import parse_package" failed: No module named 'catkin_pkg'
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python3.6
  "/opt/ros/melodic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/melodic/share/catkin/cmake/../package.xml"
  "/home/pullu/catkin_ws/build/flightlib/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/all.cmake:168 (_catkin_package_xml)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  /home/pullu/catkin_ws/devel/share/catkin_simple/cmake/catkin_simpleConfig.cmake:197 (find_package)
  cmake/catkin.cmake:2 (find_package)
  CMakeLists.txt:180 (include)

cd /home/pullu/catkin_ws/build/flightlib; catkin build --get-env flightlib | catkin env -si  /usr/bin/cmake /home/pullu/catkin_ws/src/flightmare/flightlib --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/pullu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/pullu/catkin_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
....................................................................................................
Failed     << flightlib:cmake                               [ Exited with code 1 ]                  
Failed    <<< flightlib                                     [ 1.3 seconds ]                         
Abandoned <<< flightrender                                  [ Unrelated job failed ]                
Abandoned <<< quad_launch_files                             [ Unrelated job failed ]                
Abandoned <<< quadrotor_msgs                                [ Unrelated job failed ]                
Abandoned <<< rpg_single_board_io                           [ Unrelated job failed ]                
Abandoned <<< mav_planning_msgs                             [ Unrelated job failed ]                
Abandoned <<< rotors_control                                [ Unrelated job failed ]                
Abandoned <<< rotors_hil_interface                          [ Unrelated job failed ]                
Abandoned <<< rotors_joy_interface                          [ Unrelated job failed ]                
Abandoned <<< quadrotor_common                              [ Unrelated job failed ]                
Abandoned <<< polynomial_trajectories                       [ Unrelated job failed ]                
Abandoned <<< position_controller                           [ Unrelated job failed ]                
Abandoned <<< rotors_gazebo_plugins                         [ Unrelated job failed ]                
Abandoned <<< rotors_gazebo                                 [ Unrelated job failed ]                
Abandoned <<< rpg_rotors_interface                          [ Unrelated job failed ]                
Abandoned <<< rqt_quad_gui                                  [ Unrelated job failed ]                
Abandoned <<< sbus_bridge                                   [ Unrelated job failed ]                
Abandoned <<< manual_flight_assistant                       [ Unrelated job failed ]                
Abandoned <<< state_predictor                               [ Unrelated job failed ]                
[... cut ...]
[build] Summary: 9 of 33 packages succeeded.                                                        
[build]   Ignored:   2 packages were skipped or are blacklisted.                                    
[build]   Warnings:  None.                                                                          
[build]   Abandoned: 23 packages were abandoned.                                                    
[build]   Failed:    1 packages failed.                                                             
[build] Runtime: 5.0 seconds total.

top1944, 2020-10-15:
I am running this on a virtual machine(ubuntu 18.04)， get the following errors.

Errors     << flightlib:cmake /home/dell/catkin_ws/logs/flightlib/build.cmake.002.log
/usr/include/eigen3
CMake Warning (dev) at /usr/src/googletest/CMakeLists.txt:3 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/src/googletest/googlemock/CMakeLists.txt:40 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/src/googletest/googletest/CMakeLists.txt:47 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

ImportError: "from catkin_pkg.package import parse_package" failed: No module named 'catkin_pkg'
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/dell/anaconda3/bin/python
  "/opt/ros/melodic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/melodic/share/catkin/cmake/../package.xml"
  "/home/dell/catkin_ws/build/flightlib/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/all.cmake:168 (_catkin_package_xml)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  /home/dell/catkin_ws/devel/share/catkin_simple/cmake/catkin_simpleConfig.cmake:197 (find_package)
  cmake/catkin.cmake:2 (find_package)
  CMakeLists.txt:180 (include)

peasant98, 2020-10-17:
Hmm... have you installed the Python package, `catkin_pkg`?

PulkitRustagi, 2020-10-19:
Yes, I have installed catkin_pkg . I tried reinstalling everything on a different virtual machine to omit system related problems, if any, but this is where I eventually land with all efforts. I'm pretty sure that I'm using correct versions of python, and have installed catkin_tools and catkin_pkg

Errors     << flightlib:cmake /home/pullu/catkin_ws/logs/flightlib/build.cmake.000.log              
/usr/include/eigen3
CMake Warning (dev) at /usr/src/googletest/CMakeLists.txt:3 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/src/googletest/googlemock/CMakeLists.txt:40 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/src/googletest/googletest/CMakeLists.txt:47 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

ImportError: "from catkin_pkg.package import parse_package" failed: No module named 'catkin_pkg'
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python3.6
  "/opt/ros/melodic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/melodic/share/catkin/cmake/../package.xml"
  "/home/pullu/catkin_ws/build/flightlib/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/all.cmake:168 (_catkin_package_xml)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  /home/pullu/catkin_ws/devel/share/catkin_simple/cmake/catkin_simpleConfig.cmake:197 (find_package)
  cmake/catkin.cmake:2 (find_package)
  CMakeLists.txt:180 (include)

cd /home/pullu/catkin_ws/build/flightlib; catkin build --get-env flightlib | catkin env -si  /usr/bin/cmake /home/pullu/catkin_ws/src/flightmare/flightlib --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/pullu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/pullu/catkin_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
....................................................................................................
Failed     << flightlib:cmake                               [ Exited with code 1 ]                  
Failed    <<< flightlib                                     [ 4.4 seconds ]                         
Abandoned <<< flightrender                                  [ Unrelated job failed ]                
Abandoned <<< quad_launch_files                             [ Unrelated job failed ]                
Abandoned <<< quadrotor_msgs                                [ Unrelated job failed ]                
Abandoned <<< rpg_single_board_io                           [ Unrelated job failed ]                
Abandoned <<< mav_planning_msgs                             [ Unrelated job failed ]                
Abandoned <<< rotors_control                                [ Unrelated job failed ]                
Abandoned <<< rotors_hil_interface                          [ Unrelated job failed ]                
Abandoned <<< rotors_joy_interface                          [ Unrelated job failed ]                
Abandoned <<< quadrotor_common                              [ Unrelated job failed ]                
Abandoned <<< polynomial_trajectories                       [ Unrelated job failed ]                
Abandoned <<< position_controller                           [ Unrelated job failed ]                
Abandoned <<< rotors_gazebo_plugins                         [ Unrelated job failed ]                
Abandoned <<< rotors_gazebo                                 [ Unrelated job failed ]                
Abandoned <<< rpg_rotors_interface                          [ Unrelated job failed ]                
Abandoned <<< rqt_quad_gui                                  [ Unrelated job failed ]                
Abandoned <<< sbus_bridge                                   [ Unrelated job failed ]                
Abandoned <<< manual_flight_assistant                       [ Unrelated job failed ]                
Abandoned <<< state_predictor                               [ Unrelated job failed ]                
Abandoned <<< trajectory_generation_helper                  [ Unrelated job failed ]                
[... cut ...]
[build]   Ignored:   2 packages were skipped or are blacklisted.                                    
[build]   Warnings:  None.                                                                          
[build]   Abandoned: 23 packages were abandoned.                                                    
[build]   Failed:    1 packages failed.                                                             
[build] Runtime: 48.5 seconds total.                                                                
[build] Note: Workspace packages have changed, please re-source setup files to use them.

Reka-Berci-Hajnovics, 2020-10-22:
Try to use `catkin_make` instead of `catkin build`.  For me, it solved this issue.
https://github.com/uzh-rpg/flightmare/issues/35

PulkitRustagi, 2020-10-24:
> Try to use `catkin_make` instead of `catkin build`. For me, it solved this issue.
> #35

Thanks, Worked for me too!

closed 2020-10-24 by PulkitRustagi