uzh-rpg/flightmare #39: Segmentation fault (core dumped) error in Ubuntu 20.04

arsimone, 2020-10-13:
If you compile flightlib using default gcc version 9.3 provided with Ubuntu 20.04 it gives this error. To make it work it needs to be compiled using gcc 7 that need to be manually selected. An example procedure is given [here](https://stackoverflow.com/questions/7842770/how-to-use-pip-install-where-one-requirement-has-dependency-on-version-on-gcc-4).

BananaGorilla98, 2020-11-02:
Hi. I am facing the same problem when running `python3 run_drone_control.py --train 1` in `flightrl/examples`. I tried the suggested solution by re-installing the `tensorflow-gpu==1.14` using `gcc 7` and it does not work. May I know if there is any alternative solution to make it work?

arsimone, 2020-11-02:
You do not have to install tensorflow with gcc 7, you have to install flightlib with gcc 7 and g++7. The tensorflow version is specified in the requirements.txt and is 1.8.0 not 1.14

BananaGorilla98, 2020-11-03:
Ok. Got it. Thanks!

goudon123, 2021-02-17:
Hi I have installed the flighlib using gcc version 7.5.0 and tensorflow version I used is 1.8.0.

I am able to run the training for certain steps and it show **Segmentation fault (core dumped)** .

May I know what problem caused this error?

arsimone, 2021-02-18:
Did you also change g++?
How many steps?