uzh-rpg/flightmare #46: Unable to understand PCL data extraction

Stavya1993, 2020-10-26:
Hi, I followed the example code - unity_bridge in tests on other issues, but am still not clear on how can I create a node that subscribes and receives pcl data. I am trying to build a global path planner system. Any direction would be much appreciated. Thanks and sorry for multiple new issues!

vanshilshah97, 2020-10-26:
Facing same issue with a time constrained project 
A humble request to solve as soon as possible

slimeth, 2020-10-26:
@Stavya1993 @vanshilshah97 I hope I can help you. Let me know if I did not correctly understand your problem.

For your node, you will need access to an active unity_bridge. Active means, there is a quad in the scene and it's connected.

If that's the case, you can send a **PointCloudMessage_t** which lets you configure the pointcloud. You can define the size, location, resolution and the saving path and filename. The pointcloud is saved in .ply format. https://github.com/uzh-rpg/flightmare/blob/master/flightlib/include/flightlib/bridges/unity_message_types.hpp#L119-L126

When you have configured the pointcloud you can send the message with the unity_bridge.

> unity_bridge.getPointCloud(pointcloud_msg, timeout_seconds))

It returns true if the pointcloud has been successfully saved within the timeout period.

That should be all.

vanshilshah97, 2020-10-26:
thanks for such a swift reply. I understood that. If i need to access the x,y,z of a point in cloud without saving that as a ply file. How would I proceed further. My end goal is to publish the pointcloud message on ros node ? I think I would need value of  co-ordinates for each point for that 
Or could you suggest a more efficient way ? .

slimeth, 2020-10-26:
@vanshilshah97 The pointcloud is saved on the server-side (Unity.  getPointCloud() submits the message to Unity and waits until the ply file is saved.

To achieve what you would like to have, the behaviour of the server-side would need to be altered that it returns a message here instead of saving the pointcloud. https://github.com/uzh-rpg/rpg_flightmare_unity/blob/master/Assets/Flightmare/Flightmare/Scripts/SavePointCloud.cs#L81

Stavya1993, 2020-10-27:
@slimeth Hi, thanks for the quick reply. The previous link  https://github.com/uzh-rpg/rpg_flightmare_unity/blob/master/Assets/Flightmare/Flightmare/Scripts/SavePointCloud.cs#L81 seems to be broken! Can you  share the latest link?

slimeth, 2020-10-27:
@Stavya1993 sorry for the inconvenience. The link is correct, but we currently fixing some issues privately and the repository will be public again as soon as possible.

closed 2020-10-28 by Stavya1993