uzh-rpg/flightmare #49: Import error when python3 run_drone_control.py --train 1 in Ubuntu 20.04

BananaGorilla98, 2020-11-03:
Hi.

I am trying to run the fligtmare with `python3 run_drone_control.py --train 1` in `flightmare/flightrl/examples`. However, I got the import error. The detail of the error is here: 
[image]

I am trying to look for cuda-9 in my `/usr/local` but it is nowhere. 

May I know how to solve this problem?

yun-long, 2020-11-03:
1. Make sure you followed this installation guide. In particular, was `cd flightlib && pip install .` success or not?
2. If you use `conda`, make sure you activate the conda environment.  `conda activate <env_name>`

BananaGorilla98, 2020-11-03:
Yes. I followed the installation guide and the conda environment is activated. The `cd flightlib && pip install .` success too.
However, the problem is still there.

yun-long, 2020-11-03:
Ok, I didn't read the error message. I misunderstood the problem. 
This is a TensorFlow or cuda problem. 

1. Do you have NVIDIA GPU on your laptop, if yes, follow the Tensorflow website on how to install the tensorflow-gpu properly. 
2. If no, simply install tensorflow without GPU support, `pip install tensorflow`

peasant98, 2020-11-03:
Have you installed cuda 9 @BananaGorilla98?

BananaGorilla98, 2020-11-04:
@yun-long Thanks for the suggested method. Would give a try to that solution and get back to you.
@peasant98 I don't have cuda 9 in my Ubuntu 20.04. Looks like it is Cuda 9 does not support on Ubuntu 20.04. Could other version of Cuda works?

BananaGorilla98, 2020-11-05:
@yun-long Thanks for the efforts to help! I have sorted out the solution . :) Thank you once again.

closed 2020-11-05 by BananaGorilla98