uzh-rpg/flightmare #50: Catkin build failed during flightros:make

vkotaru, 2020-11-03:
System: Ubuntu 18.04
Installation with ROS failed. 

Error 1: 
```
...src/flightmare/flightlib/src/dynamics/quadrotor_dynamics.cpp:95:10: error: ‘clamp’ is not a member of ‘std’
   return std::clamp(thrust, thrust_min_, thrust_max_);

```
Replaced it `std::clamp(thrust, thrust_min_, thrust_max_)` with `std::min(thrust_max_, std::max(thrust_min_, thrust))` temporarily to fix this issue. 

Error 2: 
```
Errors     << flightros:make .../logs/flightros/build.make.003.log
.../devel/lib/libflightlib.so: undefined reference to `flightlib::Logger::YELLOW'
.../devel/lib/libflightlib.so: undefined reference to `flightlib::Logger::WARN'
.../devel/lib/libflightlib.so: undefined reference to `flightlib::Logger::INFO'
.../devel/lib/libflightlib.so: undefined reference to `flightlib::Logger::RED'
.../devel/lib/libflightlib.so: undefined reference to `flightlib::Logger::ERROR'
.../devel/lib/libflightlib.so: undefined reference to `flightlib::Logger::RESET'
collect2: error: ld returned 1 exit status
```

Any help is appreciated. Thank you.

yun-long, 2020-11-06:
could you check which gcc/g++ version you are using? I am using 7.3.0, and everything works fine, 

the second error is a linker error because it cannot find the Logger. I am not sure why this error exists.

vkotaru, 2020-11-16:
It works now. Started everything from scratch by pulling the latest version.

closed 2020-11-16 by vkotaru