uzh-rpg/flightmare #51: Depth image rendering format results in information loss

Stavya1993, 2020-11-06:
Hello,

In using the simulators depth image rendering pipeline. I noticed that by default the images are stored in 8UC3 format. Working on the pipeline to control using depth images, I realised most depth cameras work with 16 bit images. There also seems to some sense of information loss with 8 bits when I integrate it with depth cloud to point cloud nodes. So I wanted to understand if there was any particular reason why the images rendered are in 8 bit format, or if there is any way we can already get them in 16 bit format from unity. 

Regards

slimeth, 2020-11-10:
We briefly looked into this as well. In flightmare_unity, we tried replacing the [TextureFormat](https://docs.unity3d.com/ScriptReference/TextureFormat.html) of the [texture2D](https://docs.unity3d.com/ScriptReference/Texture2D.html) with the format [RHalf](https://docs.unity3d.com/ScriptReference/TextureFormat.RHalf.html), which should provide 16-bit floating-point values. 
The change is necessary for the script RPGImageSynthesis.cs in the function getRawImage() ([similar code example](https://bitbucket.org/Unity-Technologies/ml-imagesynthesis/src/2b2bce9c0fc3dc450ead31fb17b7e3fe94510603/Assets/ImageSynthesis/ImageSynthesis.cs#lines-203:240)). We could then receive it as a CV_16UC1 on the cpp side in the script [unity_bridge.cpp in handleOutput](https://github.com/uzh-rpg/flightmare/blob/master/flightlib/src/bridges/unity_bridge.cpp#L219-L238). Unfortunately, we did not successfully receive floating-point values.
It's possible that on the unity side [Texture2D.GetRawTextureData](https://docs.unity3d.com/ScriptReference/Texture2D.GetRawTextureData.html) does behave differently as we wish.

Unfortunately, this has currently a lower priority than other tasks for Flightmare from our side. We would be glad about a contribution from you if you would be able to look into this yourself as soon as flightmare_unity is online.

Stavya1993, 2020-11-11:
I can certainly give it a shot once I have access @slimeth

negre, 2020-11-23:
I,
I worked on this problem as I'm also interested having an accurate depth map.

To solve the problem, I used a RFloat texture format in flightmare_unity side, and decode the image in a Float array, in the unity_bridge.
I also changed the "UberReplacement" shader in order to have a linear relation between the encoded depth and the real depth.

Here the diffs : 

In HuberReplacement.shader : 
```
--- a/Assets/Flightmare/ImageSynthesis/Shaders/UberReplacement.shader
+++ b/Assets/Flightmare/ImageSynthesis/Shaders/UberReplacement.shader
@@ -63,6 +63,11 @@ float4 Output(float depth01, float3 normal)
                float3 c = normal * 0.5 + 0.5;
                return float4(c, 1);
        }
+       else if (_OutputMode == 5) // DepthRaw
+       {
+               return depth01;
+       }
+
 
        // unsupported _OutputMode
        return float4(1, 0.5, 0.5, 1);
```
In RPGImageSynthesis.cs :
```--- a/Assets/Flightmare/Scripts/RPGImageSynthesis.cs
+++ b/Assets/Flightmare/Scripts/RPGImageSynthesis.cs
@@ -35,7 +35,8 @@ namespace RPGFlightmare
       CatergoryId = 1,
       DepthCompressed = 2,
       DepthMultichannel = 3,
-      Normals = 4
+      Normals = 4,
+      DepthRaw = 5,
     };
     public Dictionary<string, bool> support_antialiasing = new Dictionary<string, bool>() { };
     public Dictionary<string, bool> needs_rescale = new Dictionary<string, bool>() { };
@@ -116,7 +117,8 @@ namespace RPGFlightmare
           break;
         // Setup compressed depth for the camera
         case "depth":
-          SetupCameraWithReplacementShader(subcam, uberReplacementShader, ReplacelementModes.DepthCompressed, Color.white);
+          //SetupCameraWithReplacementShader(subcam, uberReplacementShader, ReplacelementModes.DepthCompressed, Color.white);
+          SetupCameraWithReplacementShader(subcam, uberReplacementShader, ReplacelementModes.DepthRaw, Color.white);
           if (!support_antialiasing.ContainsKey(image_mode)) support_antialiasing[image_mode] = true;
           if (!needs_rescale.ContainsKey(image_mode)) needs_rescale[image_mode] = false;
           break;
@@ -148,7 +150,7 @@ namespace RPGFlightmare
       bool supportsAntialiasing = support_antialiasing[image_mode];
       bool needsRescale = needs_rescale[image_mode];
       var depth = 24;
-      var format = RenderTextureFormat.Default;
+      var format = image_mode=="depth" ? RenderTextureFormat.RFloat : RenderTextureFormat.Default;
       var readWrite = RenderTextureReadWrite.Default;
       var antiAliasing = (supportsAntialiasing) ? Mathf.Max(1, QualitySettings.antiAliasing) : 1;
 
@@ -158,7 +160,7 @@ namespace RPGFlightmare
 
       var renderRT = (!needsRescale) ? finalRT :
           RenderTexture.GetTemporary(subcam.pixelWidth, subcam.pixelHeight, depth, format, readWrite, antiAliasing);
-      var tex = new Texture2D(width, height, TextureFormat.RGB24, false);
+      var tex = new Texture2D(width, height, image_mode=="depth" ? TextureFormat.RFloat : TextureFormat.RGB24, false);
 
       var prevActiveRT = RenderTexture.active;
       var prevCameraRT = subcam.targetTexture;
@@ -195,4 +197,4 @@ namespace RPGFlightmare
       return raw;
     }
   }
-}
\ No newline at end of file
+}
```
In unity_bridge : 
```
--- a/flightlib/src/bridges/unity_bridge.cpp
+++ b/flightlib/src/bridges/unity_bridge.cpp
@@ -216,7 +216,34 @@ bool UnityBridge::handleOutput() {
       for (size_t layer_idx = 0; layer_idx <= cam.enabled_layers.size();
            layer_idx++) {
         if (!layer_idx == 0 && !cam.enabled_layers[layer_idx - 1]) continue;
-        uint32_t image_len = cam.width * cam.height * cam.channels;
+
+       if(layer_idx==1)
+       {
+         // depth
+         uint32_t image_len = cam.width * cam.height * 4;
+         // Get raw image bytes from ZMQ message.
+         // WARNING: This is a zero-copy operation that also casts the input to
+         // an array of unit8_t. when the message is deleted, this pointer is
+         // also dereferenced.
+         const uint8_t* image_data;
+         msg.get(image_data, image_i);
+         image_i = image_i + 1;
+         // Pack image into cv::Mat
+         cv::Mat new_image =
+             cv::Mat(cam.height, cam.width, CV_32FC1);
+         memcpy(new_image.data, image_data, image_len);
+         // Flip image since OpenCV origin is upper left, but Unity's is lower
+         // left.
+         new_image = new_image * (100.f);
+         cv::flip(new_image, new_image, 0);
+         
+         
+         unity_quadrotors_[idx]->getCameras()[cam.output_index]->feedImageQueue(
+             layer_idx, new_image);
+               
+               
+       }else{
+       uint32_t image_len = cam.width * cam.height * cam.channels;
         // Get raw image bytes from ZMQ message.
         // WARNING: This is a zero-copy operation that also casts the input to
         // an array of unit8_t. when the message is deleted, this pointer is
@@ -238,6 +265,7 @@ bool UnityBridge::handleOutput() {
         }
         unity_quadrotors_[idx]->getCameras()[cam.output_index]->feedImageQueue(
           layer_idx, new_image);
+       }
       }
     }
   }
~
```

If you use ROS, you also need to modify the image format fot cv_bridge : 
```
sensor_msgs::ImagePtr depth_msg =
      cv_bridge::CvImage(std_msgs::Header(), "32FC1", img).toImageMsg();
```

It would be nice to put this modifications (or similar changes) in the project !!

slimeth, 2020-11-23:
@negre Interesting! I will check it out immediately. I tried something similar but failed to make it properly work.
I'm not yet convinced by the change you made in the Uberreplacement shader. Did you check that the depth is linear by checking the pixel values? Because the return value pow(2.2) was chosen to specifically make the depth linear. The reason [here](https://forum.unity.com/threads/gamma-space-and-linear-space-with-shader.243793/). Just to verify, your Unity color space settings is linear and not gamma?

negre, 2020-11-23:
@slimeth Yes I check the depth value at different distances and the values was clearly not linear. With the "raw" depth value, I constated a ratio on 100 between simulated and real values.

negre, 2020-11-23:
@slimeth How can I check the Unity color space ? I have not modified it...

slimeth, 2020-11-23:
@negre https://docs.unity3d.com/Manual/LinearRendering-LinearOrGammaWorkflow.html
It's in the build settings

negre, 2020-11-23:
@slimeth OK I was in "linear", that could explain the difference...

slimeth, 2020-11-23:
@negre If it was linear than it was correct. Because "Linear" is the right color space setting. I will also test it myself.

slimeth, 2020-11-24:
@negre Did you try to save an image? I tried to reproduce your code. The output was a floating-point value, but the output in rviz and the saved image did not make much sense.
[image]

negre, 2020-11-24:
@slimeth Here a screen capture in the industrial environment (displayed with rqt_image_view) :
[image]

I have some trouble with particles systems, but the depth is nice.

Are you sure to correctly set the right format to cv_bridge or in to correcly copy the stream in unity_bridge ?

slimeth, 2020-11-24:
@negre Use the[ standard Unity particle shader](https://docs.unity3d.com/Manual/shader-StandardParticleShaders.html). This environment seems to have some other strange particle shader.

And I basically copy-pasted your code and this lead to this weird output.

negre, 2020-11-24:
@slimeth That's strange !
Maybe the RFloat texture+renderTexture format is not well supported on all hardwares, do you use a nvidia card ?
It is possible also to use the "DepthMultichannel" to render the depth in R+G channels and decode this as uint16_t in unity_bridge...

I will also check the particle shader...

slimeth, 2020-11-24:
@negre I'm using a Nvidia GeForce GTX 1050i graphic card. 

I just try to have the same setup as you and see if it's some hardware related problem. Maybe you could make a pull request for review with the setup that works for you. Unfortunately, I will only have time to review it before next Monday.

negre, 2020-11-25:
@slimeth I have submitted a pull request in the two repositories.
I observed the same image than you when I launched the new version of flightmare with the non patched flightmare_unity, are you sure you ran the modified unity environment ?

antonilo, 2020-11-25:
also works for me, thanks @negre!

Stavya1993, 2020-11-25:
Have been wanting to try this out for sometime now. Will look into it as well when I get time next week. Thank @negre  for the great job!

slimeth, 2020-11-30:
> @slimeth I have submitted a pull request in the two repositories.
> I observed the same image than you when I launched the new version of flightmare with the non patched flightmare_unity, are you sure you ran the modified unity environment?

@negre you were right. What a silly mistake by me. Accidentally an unpatched flightmare_unity from another catkin workspace was used. Everything works like a charm! Good job!

geri03, 2020-11-30:
@negre I have downloaded the Unity-project from your repository. After building it with the unity editor, I cannot connect to unity when executing the camera.launch file. The flightmare-unity starts, but the connection fails.
Any hints on that?

negre, 2020-11-30:
@geri03 Actually I had a bug in the file "sceneSchedule.cs", I didn't commit my change because it is not relative to the depth, but maybe you have the same bug. I have commited my change now, you can try to update and restart. If it doesn't work, you can try to build with "Development Build" mode to see the error.