uzh-rpg/flightmare #52: Building components into the warehouse environment of unity

Stavya1993, 2020-11-06:
I believe the rpg_flightmare_unity repository is responsible for this. Can i get some idea on if its still possible to do it without it and how I can do it. Alternatively, can I get an idea when the repository should be up again?

slimeth, 2020-11-09:
@Stavya1993 Could you clarify, what exactly you mean by components in this context. However, if you want to change anything in a scene, you will need to do it within rpg_flightmare_unity. The rpg_flightmare_unity respectively then newly called flightmare_unity will be public as soon as the pull request for dev/samples is merged in this repository. We hope that's soon the case!
Just a warning: The warehouse will be replaced by this [environment](https://assetstore.unity.com/packages/3d/environments/industrial/rpg-fps-game-assets-for-pc-mobile-industrial-set-v2-0-86679).
In the documentation which is in process, we will explain how you can add your environments to flightmare_unity. If you purchase the [warehouse](https://assetstore.unity.com/packages/3d/environments/industrial/hangar-building-modular-142104) you can modify the components in the warehouse. All scenes previously available in the binary will be still in there, just the environment for the public repository had to be replaced.

Stavya1993, 2020-11-10:
@slimeth, by components I meant to add elements into the scene to test my path planning system. Eagerly waiting for the new flightmare_unity to test things out! Thanks!

closed 2020-11-10 by Stavya1993