uzh-rpg/flightmare #55: subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.

guytenn, 2020-11-10:
I'm getting this error when calling pip install . in the flightlib directory (after installing all prerequisites). Any ideas?
Full log:

ERROR: Command errored out with exit status 1:
     command: /home/guy/Projects/venv/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/guy/Projects/flightmare/flightlib/setup.py'"'"'; __file__='"'"'/home/guy/Projects/flightmare/flightlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps
         cwd: /home/guy/Projects/flightmare/flightlib/
    Complete output (162 lines):
    running develop
    running egg_info
    writing flightgym.egg-info/PKG-INFO
    writing dependency_links to flightgym.egg-info/dependency_links.txt
    writing requirements to flightgym.egg-info/requires.txt
    writing top-level names to flightgym.egg-info/top_level.txt
    reading manifest file 'flightgym.egg-info/SOURCES.txt'
    writing manifest file 'flightgym.egg-info/SOURCES.txt'
    running build_ext
    -- The C compiler identification is GNU 9.3.0
    -- The CXX compiler identification is GNU 9.3.0
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- ======================  !Flightmare!  ======================
    -- ======> Setup Dependencies
    -- No sufficient Eigen version (3.3.4) found.
    -- Restoring to download Eigen sources.
    -- Getting Eigen...
    Cloning into 'eigen3'...
    Already on 'master'
    -- Eigen downloaded!
    -- Eigen3 include dir:
    -- Getting Pybind11...
    Cloning into 'pybind11-src'...
    Already on 'master'
    -- Pybind11 downloaded!
    -- pybind11 v2.6.1 dev1
    -- Found PythonInterp: /home/guy/Projects/venv/bin/python3 (found version "3.8.2")
    -- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so
    -- Performing Test HAS_FLTO
    -- Performing Test HAS_FLTO - Success
    -- Getting yaml-cpp...
    -- Yaml downloaded!
    -- Found OpenCV: /usr (found version "4.2.0")
    -- Found OpenMP_C: -fopenmp (found version "4.5")
    -- Found OpenMP_CXX: -fopenmp (found version "4.5")
    -- Found OpenMP: TRUE (found version "4.5")
    INFOBuild time could be improved with ccache!
    INFO    sudo apt install ccache
    -- ======> Setup Compilation
    -- Enabling fast optimization flags!
    -- The activated CXX RELEASE configuration is:
     -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -DNDEBUG  -fopenmp  -Ofast  -march=native
    -- The activated CXX DEBUG configuration is:
     -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -g -fopenmp
    -- ======>  Setup Build
    -- Getting gtests...
    Cloning into 'googletest-src'...
    Already on 'master'
    -- gtests downloaded!
    -- Looking for pthread.h
    -- Looking for pthread.h - found
    -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[... 77 lines cut ...]
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/home/guy/Projects/venv/lib/python3.8/site-packages/setuptools/command/develop.py", line 38, in run
        self.install_for_development()
      File "/home/guy/Projects/venv/lib/python3.8/site-packages/setuptools/command/develop.py", line 140, in install_for_development
        self.run_command('build_ext')
      File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/home/guy/Projects/flightmare/flightlib/setup.py", line 59, in run
        self.build_extension(ext)
      File "/home/guy/Projects/flightmare/flightlib/setup.py", line 90, in build_extension
        subprocess.check_call(['cmake', '--build', '.'] +
      File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
    ----------------------------------------
  ERROR: Can't roll back flightgym; was not uninstalled
ERROR: Command errored out with exit status 1: /home/guy/Projects/venv/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/guy/Projects/flightmare/flightlib/setup.py'"'"'; __file__='"'"'/home/guy/Projects/flightmare/flightlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output

slimeth, 2020-11-10:
It's a version problem as we don't use CV 4.2. There has been a solution proposed in [this issue](https://github.com/uzh-rpg/flightmare/issues/23).

siddas27, 2021-04-12:
Following the below link worked for me.
[https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip](https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip)