uzh-rpg/flightmare #56: Depth image problem in the forest environment

negre, 2020-11-12:
Hello,
In the forest environment, the ground seems to be removed in the depth image as we can see on this image :
[image]

Is it possible to correct this problem (in Unity I suppose) ?

slimeth, 2020-11-16:
@negre We know the problem and how to fix it. Which release version of the binaries do you use?

negre, 2020-11-17:
I use the version 0.0.3 (Sept 22) of RPG_Flightmare with ROS.

yun-long, 2020-11-18:
we release a new version, please download the newest version. (0.0.4)
Let us know if you still encounter this problem.

negre, 2020-11-19:
Great !
I test with the new release and it works well.

Thank-you !

bowenXuu, 2023-12-14:
Hello，
I have encountered the same problem after building in unity (also using the Nature Forest scenario), how can I solve this problem?