uzh-rpg/flightmare #57: Unity Connection time out!

ccooddeerr, 2020-11-13:
Thanks for your excellent work with the Flightmare simulator which would be a good contribution to the realm of robotics.
btw,I can't find the file rgb_flightmare_unity to make a connection when I installed unityhub

yun-long, 2020-11-16:
To use unity rendering, you need first download the binary from Releases and extract it into the flightrender folder. To enable unity for visualization, double click the extracted executable file RPG_Flightmare.x84-64 and then test a pre-trained controller

https://github.com/uzh-rpg/flightmare/releases

ccooddeerr, 2020-11-17:
Thanks for your useful advice sincerely.But following the wiki of python usage,I can't control the virtual quadrotor.