uzh-rpg/flightmare #59: Windows Support

laurelkeys, 2020-11-16:
Hi! Is a Windows relase of the UI (**RPG_Flightmare.x86_64**) on the roadmap?

Also, looking at the [**Unity Developer Guide**](https://github.com/uzh-rpg/flightmare/wiki/Unity-Developer-Guide) on the Wiki suggests that any Unity-related code (e.g. the rendering engine and also the C# scripts mentioned in the paper) will soon be made available. 

Do you have any prediction of when `rpg_flightmare_unity` should be released then? As this would most likely allow us to build a Windows executable ourselves.

yun-long, 2020-11-16:
We don't have a plan for a Windows Release but will think about it since you are requesting it. 

Approximately, we will release rpg_flightmare_unity this week.

yun-long, 2020-11-18:
We just release the flightmare unity and the documentation, you can check it out. 

https://github.com/uzh-rpg/flightmare_unity

https://flightmare.readthedocs.io/en/latest/

laurelkeys, 2020-11-18:
> We just release the flightmare unity and the documentation, you can check it out.

Awesome! Thanks @yun-long

closed 2020-11-18 by laurelkeys