uzh-rpg/flightmare #65: problem with flightmare

valeria025, 2020-11-27:
I can't download the simulator on my pc. I use windows operation system. I tried to download virtual box,but it not works.
so i tried to download ubuntu 20.10 on my pc, but i can't install Flightmare. 
Someone can help me, because i need to use it, for my Internship for my univercity. 
Thanks

peasant98, 2020-11-27:
@valeria025 Can you give a more detailed description of the problem?

valeria025, 2020-11-27:
I tried to download the script on Ubuntu 20.10, but i can't open it

sansha, 2020-12-11:
Did you follow this [guide](https://flightmare.readthedocs.io/en/latest/getting_started/quick_start.html#)? where did you receive what kind of error?

valeria025, 2020-12-26:
hi how can install flightmare on unity hub. Because when i create the
project, i don't have the screen of flightmare, but i have the scene that i
import.