uzh-rpg/flightmare #67: Collision with environment?

Hersh500, 2020-11-28:
Does the Unity side detect collisions between the drone and environment objects? I'm directly printing out the collision flag from unity_bridge.cpp and manually driving the quad into an object in the warehouse scene, but it's always false.

slimeth, 2020-12-01:
https://github.com/uzh-rpg/flightmare/releases/tag/0.0.5 The new standalone fixes it. The colliders in the scene where not triggers by default and were therefore not detected. Let me know if it works for you! @Hersh500

Hersh500, 2020-12-01:
Thanks for the response! I downloaded the new binary and pulled the new unity bridge code, but now I'm getting an error where the bridge seems to be unable to connect to the standalone, even though the standalone is getting launched.

slimeth, 2020-12-01:
@Hersh500 could you please post the error message. That's strange, for me it works to connect to the examples in flightros. Do these work for you? (Try to launch them with the option roslaunch camera.launch use_unity_editor:=1 when you have already the new standalone running). Is it possible that you have still somehow an obsolete flighmare in flightrender that gets launched? Did you try cleaning your catkin workspace and re-build and re-source it?

shuaikangma, 2022-03-18:
> https://github.com/uzh-rpg/flightmare/releases/tag/0.0.5 The new standalone fixes it. The colliders in the scene where not triggers by default and were therefore not detected. Let me know if it works for you! @Hersh500

I use the new binary to test collision in ros(C++) with code:
`if(quad_ptr->getCollision()){
                printf("-----------colission----------------\n");
            }
            else{
                printf("no colission\n");
            }`
but it always print no colission, even if the drone go through the boader of the warehouse:
[image]
The `has_collision_check` in ~/flightmare_ws/src/flightmare/flightlib/include/flightlib/bridges/unity_message_types.hpp was set to be true. Could you tell me where should modify to get colission dector work?
thanks!

Lu-tju, 2022-06-09:
> 
I also encountered the same collision detection problem. Have you solved it?  Thanks！

slimeth, 2022-06-20:
@Lu-tju did you try another environment? E.g. Industrial. It could be that this environment does not have the colliders properly configured in the unity scene.