uzh-rpg/flightmare #68: Issues installing on Ubuntu

guytenn, 2020-11-29:
I've been having trouble installing flightmare. I've attempted installing it on fresh installations of Ubuntu 20 on 4 different machines. I always get the same error:

(venv) guy@guy-desktop:~/Projects/flightmare/flightlib$ pip install .
Processing /home/guy/Projects/flightmare/flightlib
Building wheels for collected packages: flightgym
  Building wheel for flightgym (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/guy/Projects/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-wkhfi20y/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-wkhfi20y/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-cwah6dfg
       cwd: /tmp/pip-req-build-wkhfi20y/
  Complete output (104 lines):
  running bdist_wheel
  running build
  running build_ext
  -- The C compiler identification is GNU 9.3.0
  -- The CXX compiler identification is GNU 9.3.0
  -- Check for working C compiler: /usr/bin/cc
  -- Check for working C compiler: /usr/bin/cc -- works
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Check for working CXX compiler: /usr/bin/c++
  -- Check for working CXX compiler: /usr/bin/c++ -- works
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- ======================  !Flightmare!  ======================
  -- ======> Setup Dependencies
  -- No sufficient Eigen version (3.3.4) found.
  -- Restoring to download Eigen sources.
  -- Getting Eigen...
  Cloning into 'eigen3'...
  Note: switching to '3.3.4'.
  
  You are in 'detached HEAD' state. You can look around, make experimental
  changes and commit them, and you can discard any commits you make in this
  state without impacting any branches by switching back to a branch.
  
  If you want to create a new branch to retain commits you create, you may
  do so (now or later) by using -c with the switch command. Example:
  
    git switch -c <new-branch-name>
  
  Or undo this operation with:
  
    git switch -
  
  Turn off this advice by setting config variable advice.detachedHead to false
  
  HEAD is now at 3dc3a0ea2 bump to 3.3.4
  -- Eigen downloaded!
  -- Eigen3 include dir:
  -- Getting Pybind11...
  Cloning into 'pybind11-src'...
  Already on 'master'
  -- Pybind11 downloaded!
  -- pybind11 v2.6.2 dev1
  -- Found PythonInterp: /home/guy/Projects/venv/bin/python3 (found version "3.8.5")
  -- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so
  -- Performing Test HAS_FLTO
  -- Performing Test HAS_FLTO - Success
  -- Getting yaml-cpp...
  -- Yaml downloaded!
  CMake Error at CMakeLists.txt:50 (find_package):
    By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
    asked CMake to find a package configuration file provided by "OpenCV", but
    CMake did not find one.
  
    Could not find a package configuration file provided by "OpenCV" with any
[... 116 lines cut ...]
      File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/home/guy/Projects/venv/lib/python3.8/site-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.8/distutils/command/install.py", line 589, in run
        self.run_command('build')
      File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3.8/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-req-build-wkhfi20y/setup.py", line 59, in run
        self.build_extension(ext)
      File "/tmp/pip-req-build-wkhfi20y/setup.py", line 88, in build_extension
        subprocess.check_call(['cmake', ext.sourcedir] +
      File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-req-build-wkhfi20y', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-req-build-wkhfi20y/build/lib.linux-x86_64-3.8/', '-DPYTHON_EXECUTABLE=/home/guy/Projects/venv/bin/python3', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/guy/Projects/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-wkhfi20y/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-wkhfi20y/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-fb6tumbe/install-record.txt --single-version-externally-managed --compile --install-headers /home/guy/Projects/venv/include/site/python3.8/flightgym Check the logs for full command output.

peasant98, 2020-11-29:
Hi, do you have OpenCV installed? I had a similar error.

guytenn, 2020-11-30:
I have opencv installed. It feels like a gcc version problem to me but I'm not sure