uzh-rpg/flightmare #71: issues installing flightlib for Ubuntu 18.04

mihirk284, 2020-12-02:
Hello, 
I am having some trouble installing flightmare for Ubuntu 18.04. I have been able to follow the instructions up to [this one](https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip#install-flighmare-flightlib). However, the command `pip install .` is giving out an error. I am pasting the log below:-
```
Processing /home/mihir/rl/flightmare/flightlib
Building wheels for collected packages: flightgym
  Building wheel for flightgym (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/mihir/miniconda3/envs/flightmare/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-tt9h9txj/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-tt9h9txj/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-6idsieyc
       cwd: /tmp/pip-req-build-tt9h9txj/
  Complete output (190 lines):
  running bdist_wheel
  running build
  running build_ext
  -- The C compiler identification is GNU 8.4.0
  -- The CXX compiler identification is GNU 8.4.0
  -- Check for working C compiler: /usr/bin/cc
  -- Check for working C compiler: /usr/bin/cc -- works
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Check for working CXX compiler: /usr/bin/c++
  -- Check for working CXX compiler: /usr/bin/c++ -- works
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- ======================  !Flightmare!  ======================
  -- ======> Setup Dependencies
  -- Using system provided Eigen.
  /usr/include/eigen3
  -- Eigen3 include dir: /usr/include/eigen3
  -- Getting Pybind11...
  Cloning into 'pybind11-src'...
  Already on 'master'
  -- Pybind11 downloaded!
  -- pybind11 v2.6.2 dev1
  -- Found PythonInterp: /home/mihir/miniconda3/envs/flightmare/bin/python (found version "3.6.9")
  -- Found PythonLibs: /home/mihir/miniconda3/envs/flightmare/lib/libpython3.6m.so
  -- Performing Test HAS_FLTO
  -- Performing Test HAS_FLTO - Success
  -- Getting yaml-cpp...
  -- Yaml downloaded!
  -- Found OpenCV: /usr (found version "3.2.0")
  -- Found OpenMP_C: -fopenmp (found version "4.5")
  -- Found OpenMP_CXX: -fopenmp (found version "4.5")
  -- Found OpenMP: TRUE (found version "4.5")
  INFOBuild time could be improved with ccache!
  INFO    sudo apt install ccache
  -- ======> Setup Compilation
  -- Enabling fast optimization flags!
  -- The activated CXX RELEASE configuration is:
   -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -DNDEBUG  -fopenmp  -Ofast  -march=native
  -- The activated CXX DEBUG configuration is:
   -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -g -fopenmp
  -- ======>  Setup Build
  -- Getting gtests...
  Cloning into 'googletest-src'...
  Already on 'master'
  -- gtests downloaded!
  -- Looking for pthread.h
  -- Looking for pthread.h - found
  -- Looking for pthread_create
  -- Looking for pthread_create - not found
  -- Looking for pthread_create in pthreads
  -- Looking for pthread_create in pthreads - not found
  -- Looking for pthread_create in pthread
  -- Looking for pthread_create in pthread - found
  -- Found Threads: TRUE
  -- ================  !Done. No more nightmare!  ================
[... 118 lines cut ...]
    warnings.warn(msg)
  WARNING: '' not a valid package name; please use only .-separated package names in setup.py
  running bdist_wheel
  running build
  running build_py
  creating build/lib
  copying ./build/setup.py -> build/lib
  installing to build/bdist.linux-x86_64/wheel
  running install
  running install_lib
  copying build/lib/setup.py -> build/bdist.linux-x86_64/wheel
  error: could not create 'build/bdist.linux-x86_64/wheel/setup.py': No such file or directory
  ----------------------------------------
  ERROR: Failed building wheel for flightgym
  Running setup.py clean for flightgym
Failed to build flightgym
Installing collected packages: flightgym
    Running setup.py install for flightgym ... done
  DEPRECATION: flightgym was installed using the legacy 'setup.py install' method, because a wheel could not be built for it. pip 21.0 will remove support for this functionality. A possible replacement is to fix the wheel build issue reported above. You can find discussion regarding this at https://github.com/pypa/pip/issues/8368.
Successfully installed flightgym-0.0.1
```

Can someone please help me to tell me if I am going wrong somewhere with this or provide a solution to this issue?

Thanks,

mihirk284, 2020-12-02:
In addition to the above, if I just kept on going as per the instructions up to [this one](https://github.com/uzh-rpg/flightmare/wiki/Basic-Usage-with-Python#train-neural-network-controller-using-ppo), on running the command `python3 run_drone_control.py --train 1` ,I get the following error:-

```
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
[krypton:13798] *** Process received signal ***
[krypton:13798] Signal: Segmentation fault (11)
[krypton:13798] Signal code: Address not mapped (1)
[krypton:13798] Failing at address: 0x29
[krypton:13798] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x128a0)[0x7f5883f058a0]
[krypton:13798] [ 1] /lib/x86_64-linux-gnu/libc.so.6(cfree+0x3d)[0x7f5883b999fd]
[krypton:13798] [ 2] /home/mihir/miniconda3/envs/flightmare/lib/python3.6/site-packages/flightgym.cpython-36m-x86_64-linux-gnu.so(+0x385ec)[0x7f5817a475ec]
[krypton:13798] [ 3] [0x7ffc0fa21e70]
[krypton:13798] *** End of error message ***
[1]    13798 segmentation fault (core dumped)  python3 run_drone_control.py --train 1
```

yun-long, 2020-12-02:
There was a problem in the setup.py under flightmare/flightlib. 

You can pull from the master and try it again.

mihirk284, 2020-12-02:
Thank you @yun-long   after pulling the new master the building of wheel is resolved. However, the issue mentioned in the second comment on this thread still persists.
Is there anything I may be doing incorrectly?

yun-long, 2020-12-02:
hmmmm, it is difficult to inspect why it happens. 

Could you do the following:
1) build and compile the project without using pip `cd flightmare/flightlib/build && cmake .. && make -j10`
2) run the unit tests by `./test_lib `

do all unit tests passed? if no, what are the output?

if yes, try to create a new conda environment and reinstall flightmare and run the example again.

mihirk284, 2020-12-02:
@yun-long  Thank you for your quick response
I get a segmentation fault while running the test
the output is as follows:-
```
Running main() from /home/mihir/rl/flightmare/flightlib/externals/googletest-src/googletest/src/gtest_main.cc
[==========] Running 28 tests from 8 test suites.
[----------] Global test environment set-up.
[----------] 4 tests from QuadrotorDynamics
[ RUN      ] QuadrotorDynamics.Constructor
Quadrotor Dynamics:
mass =             [1.2]
arm_l =            [0.25]
t_BM =             [ 0.177 -0.177 -0.177  0.177]
                   [-0.177 -0.177  0.177  0.177]
                   [0 0 0 0]
inertia =          [0.0281      0      0]
                   [     0 0.0281      0]
                   [     0      0 0.0437]
motor_omega_min =  [150]
motor_omega_max =  [2e+03]
motor_tau_inv =    [20]
thrust_map =       [1.33e-06  0.00384    -1.77]
kappa =            [0.016]
thrust_min =       [0]
thrust_max =       [11.2]
omega_max =        [6 6 6]

     1      1      1      1
 0.177 -0.177 -0.177  0.177
-0.177 -0.177  0.177  0.177
 0.016 -0.016  0.016 -0.016
[       OK ] QuadrotorDynamics.Constructor (0 ms)
[ RUN      ] QuadrotorDynamics.Dynamics
[       OK ] QuadrotorDynamics.Dynamics (0 ms)
[ RUN      ] QuadrotorDynamics.VectorReference
[       OK ] QuadrotorDynamics.VectorReference (1 ms)
[ RUN      ] QuadrotorDynamics.LoadParams
Quadrotor Dynamics:
mass =             [0.73]
arm_l =            [0.17]
t_BM =             [ 0.12 -0.12 -0.12  0.12]
                   [-0.12 -0.12  0.12  0.12]
                   [0 0 0 0]
inertia =          [0.00791       0       0]
                   [      0 0.00791       0]
                   [     0      0 0.0123]
motor_omega_min =  [150]
motor_omega_max =  [3e+03]
motor_tau_inv =    [1e+04]
thrust_map =       [1.33e-06  0.00384    -1.77]
kappa =            [0.016]
thrust_min =       [0]
thrust_max =       [11.2]
omega_max =        [6 6 6]

[       OK ] QuadrotorDynamics.LoadParams (0 ms)
[----------] 4 tests from QuadrotorDynamics (1 ms total)

[----------] 4 tests from Quadrotor
[ RUN      ] Quadrotor.Constructor
[       OK ] Quadrotor.Constructor (1 ms)
[ RUN      ] Quadrotor.ResetSimulator
[       OK ] Quadrotor.ResetSimulator (0 ms)
[ RUN      ] Quadrotor.RunQuadCmdFeedThrough
[       OK ] Quadrotor.RunQuadCmdFeedThrough (0 ms)
[ RUN      ] Quadrotor.RunSimulatorBodyRate
[       OK ] Quadrotor.RunSimulatorBodyRate (1 ms)
[----------] 4 tests from Quadrotor (2 ms total)

[----------] 3 tests from QuadrotorEnv
[ RUN      ] QuadrotorEnv.Constructor
[QuadrotorEnv] Environment configuration path "/home/mihir/rl/flightmare/flightlib/configs/quadrotor_env.yaml".
[1]    26478 segmentation fault (core dumped)  ./test_lib
```

yun-long, 2020-12-02:
did you do this 

Add Environment Variable
Add FLIGHTMARE_PATH environment variable to your .bashrc file:

```
echo "export FLIGHTMARE_PATH=/home/mihir/rl/flightmare" >> ~/.bashrc
source ~/.bashrc
```

and check by 
`
echo $FLIGHTMARE_PATH
`

mihirk284, 2020-12-02:
I set the environment variable
the output is echo is as expected
`echo $FLIGHTMARE_PATH` -> `/home/mihir/rl/flightmare`

I am using zsh instead of bash ( I don't know if that makes a difference)

yun-long, 2020-12-02:
I don't think so. I cannot reproduce the problem on my machine.

yun-long, 2020-12-02:
I am also using Ubuntu 18.04. The only difference I can see is that the C/CXX compiler version are different. 

-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ======================  !Flightmare!  ======================
-- ======> Setup Dependencies
-- Using system provided Eigen.
/usr/include/eigen3
-- Eigen3 include dir: /usr/include/eigen3
-- Getting Pybind11...
-- Pybind11 downloaded!
-- pybind11 v2.6.2 dev1
-- Found PythonInterp: /usr/bin/python3.6 (found version "3.6.8") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Getting yaml-cpp...
-- Yaml downloaded!
-- Found OpenCV: /usr (found version "3.2.0") 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- ======> Setup Compilation
-- Enabling fast optimization flags!
-- The activated CXX RELEASE configuration is:
  -fPIC -Wall -DNDEBUG  -fopenmp  -Ofast  -march=native 
-- The activated CXX DEBUG configuration is:
  -fPIC -Wall -g -fopenmp
-- ======>  Setup Build  
-- Getting gtests...
-- gtests downloaded!
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- ================  !Done. No more nightmare!  ================
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sysadmin/Desktop/flightmare/flightlib/build

yun-long, 2020-12-02:
[==========] Running 28 tests from 8 test suites.
[----------] Global test environment set-up.
[----------] 4 tests from QuadrotorDynamics
[ RUN      ] QuadrotorDynamics.Constructor
Quadrotor Dynamics:
mass =             [1.2]
arm_l =            [0.25]
t_BM =             [ 0.177 -0.177 -0.177  0.177]
                   [-0.177 -0.177  0.177  0.177]
                   [0 0 0 0]
inertia =          [0.0281      0      0]
                   [     0 0.0281      0]
                   [     0      0 0.0437]
motor_omega_min =  [150]
motor_omega_max =  [2e+03]
motor_tau_inv =    [20]
thrust_map =       [1.33e-06  0.00384    -1.77]
kappa =            [0.016]
thrust_min =       [0]
thrust_max =       [11.2]
omega_max =        [6 6 6]

     1      1      1      1
 0.177 -0.177 -0.177  0.177
-0.177 -0.177  0.177  0.177
 0.016 -0.016  0.016 -0.016
[       OK ] QuadrotorDynamics.Constructor (0 ms)
[ RUN      ] QuadrotorDynamics.Dynamics
[       OK ] QuadrotorDynamics.Dynamics (0 ms)
[ RUN      ] QuadrotorDynamics.VectorReference
[       OK ] QuadrotorDynamics.VectorReference (1 ms)
[ RUN      ] QuadrotorDynamics.LoadParams
Quadrotor Dynamics:
mass =             [0.73]
arm_l =            [0.17]
t_BM =             [ 0.12 -0.12 -0.12  0.12]
                   [-0.12 -0.12  0.12  0.12]
                   [0 0 0 0]
inertia =          [0.00791       0       0]
                   [      0 0.00791       0]
                   [     0      0 0.0123]
motor_omega_min =  [150]
motor_omega_max =  [3e+03]
motor_tau_inv =    [1e+04]
thrust_map =       [1.33e-06  0.00384    -1.77]
kappa =            [0.016]
thrust_min =       [0]
thrust_max =       [11.2]
omega_max =        [6 6 6]

[       OK ] QuadrotorDynamics.LoadParams (0 ms)
[----------] 4 tests from QuadrotorDynamics (1 ms total)

[----------] 4 tests from Quadrotor
[ RUN      ] Quadrotor.Constructor
[       OK ] Quadrotor.Constructor (0 ms)
[ RUN      ] Quadrotor.ResetSimulator
[       OK ] Quadrotor.ResetSimulator (0 ms)
[ RUN      ] Quadrotor.RunQuadCmdFeedThrough
[       OK ] Quadrotor.RunQuadCmdFeedThrough (1 ms)
[ RUN      ] Quadrotor.RunSimulatorBodyRate
[       OK ] Quadrotor.RunSimulatorBodyRate (0 ms)
[----------] 4 tests from Quadrotor (1 ms total)

[----------] 3 tests from QuadrotorEnv
[ RUN      ] QuadrotorEnv.Constructor
[QuadrotorEnv] Environment configuration path "/home/sysadmin/Projects/RPG_Flightmare/catkin_flightmare/src/flightmare/flightlib/configs/quadrotor_env.yaml".
Quadrotor Environment:
obs dim =            [12]
act dim =            [4]
[... 73 lines cut ...]
[Printing] mean|std:  0.111 | 0 ms    [min|max:  0.111 | 0.111 ms]

A QuadState
State at 0s: [0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]
[       OK ] Logger.SimpleLogging (0 ms)
[ RUN      ] Logger.NoColorLogging
[Test]         This is a text stream log.
[Test]         Info:    This is an info log.
[Test]         Warning: This could be a warning, but just for demo.
[Test]         Error:   This could be an error, but just for demo.
[       OK ] Logger.NoColorLogging (0 ms)
[----------] 2 tests from Logger (0 ms total)

[----------] 3 tests from QuadState
[ RUN      ] QuadState.Constructor
[       OK ] QuadState.Constructor (0 ms)
[ RUN      ] QuadState.Accessors
[       OK ] QuadState.Accessors (0 ms)
[ RUN      ] QuadState.Compare
[       OK ] QuadState.Compare (0 ms)
[----------] 3 tests from QuadState (0 ms total)

[----------] Global test environment tear-down
[==========] 28 tests from 8 test suites ran. (185 ms total)
[  PASSED  ] 28 tests.

mihirk284, 2020-12-02:
I am also stumped. 
I'll try matching the versions of the C/CXX compilers.

I noticed something different on the above outputs as well... 
for the compile comments, your files are being generated to the path `-- Build files have been written to: /home/sysadmin/Desktop/flightmare/flightlib/build`

but for running the test, another configuration path is being used:-
`[QuadrotorEnv] Environment configuration path "/home/sysadmin/Projects/RPG_Flightmare/catkin_flightmare/src/flightmare/flightlib/configs/quadrotor_env.yaml".`

Is it possible that this may be a mismatch since your quadrotor_env.yaml file should have been loaded from:- 
`"/home/sysadmin/Desktop/flightmare/flightlib/configs/quadrotor_env.yaml"`

mihirk284, 2020-12-03:
@yun-long  I matched the compiler versions and the issue is resolved. 
Thank you for your help :)

closed 2020-12-03 by mihirk284