uzh-rpg/flightmare #75: Flightmare installation Ubuntu 16.04

MiquelEspinosa, 2021-01-04:
I have been trying to install Flightmare on my **Ubuntu 16.04** machine for two days and I am unable to install it. 
I'm running on ubuntu 16, and I have installed **ROS Kinetic** with **gazebo 7**.

I think there is a dependency problem because gazebo 7 needs `protoc --version` of `2.6.1`, but at the same time, **Flightmare** needs the protobuf version of 3.0.0. 
I have followed the installation instructions and I am unable by any means to solve this problem of dependencies and versions.

Any help would be highly appreciated. Many thanks.

yun-long, 2021-01-26:
sorry for the late reply, but can you check which error messages you have. It seems like a ROS problem, not related to Flightmare.

MiquelEspinosa, 2021-01-31:
Thanks, Yun.
I installed it on Ubuntu 18.04 and it worked fine (consider adding this #35 info in the doc, as it can be useful for future users).
If I should finally use it on Ubuntu 16.04 I'll come back to this issue and keep you informed. 
Feel free to close this issue.
Many thanks.

closed 2021-04-10 by miquel-espinosa