uzh-rpg/flightmare #76: IMU integration

marcinolokk, 2021-01-13:
Hi everyone
Is it possible to get the IMU sensor data in the simulation? I would like to use VIO on top of that.
Thank!

yun-long, 2021-01-26:
yes. you would need to use the Rotors for the IMU simulation. 
check this example, if you do `rostopic list` you can see the IMU topice.

shuaikangma, 2022-03-18:
Hello, I also want to use IMU sensor data for my activate SLAM project.
But I don't know how to use the Rotors and which example I should refer to, I think you may forget to attach a link to the example.  Could you give me some guidance for how to add a IMU to the drone?
Thanks!

sandilyasg, 2023-10-05:
@marcinolokk @yun-long can you please explain how to use RotorS for the IMU simulation? Is there a link to view the example because it does not appear in @yun-long 's comment.