uzh-rpg/flightmare #79: Quadrotor Dynamics Model

pdalrymple, 2021-01-29:
The flightmare research paper mentions a flexible interface to three quadrotor dynamics.

1. Is there any documentation on how/where one sets and changes the quadrotor dynamics?
2. Are users able to import their own dynamic models?  Is so, what are the steps?

yun-long, 2021-02-07:
Hi, 
 
Unfortunately, we don't have very specific documentation about how to set or change the quadrotor dynamics. In summary, we mainly use the quadrotor for two kinds of applications, a) reinforcement learning, b) model-based control, such as MPC. For a), please refer to [quadrotor_env](https://github.com/uzh-rpg/flightmare/blob/master/flightlib/src/envs/quadrotor_env/quadrotor_env.cpp). We implemented an OpenAI gym style environment where the task is to stabilize the quadrotor from randomly initialized poses. The environment is then wrapped via pybind11, see [here](https://github.com/uzh-rpg/flightmare/blob/master/flightlib/src/wrapper/pybind_wrapper.cpp). Finally, we can python and RL algorithms to solve the problem. see [here](https://github.com/uzh-rpg/flightmare/blob/master/flightrl/examples/run_drone_control.py)
 For b) we use the RotorS to simulate the quadrotor and RPG quadrotor control framework to control the quadrotor and use Flightmare Unity Standalone to simulate 3D environment and images. please see [here ](https://github.com/uzh-rpg/flightmare/blob/master/flightros/src/pilot/flight_pilot.cpp).

mrunaljsarvaiya, 2023-11-22:
@yun-long any chance you could link some documentation/paper that the dynamics model defined in the following file is based off of ? https://github.com/uzh-rpg/flightmare/blob/master/flightlib/src/dynamics/quadrotor_dynamics.cpp?