uzh-rpg/flightmare #80: Flight lib installation problem because of pybind

HasanFarag, 2021-02-06:
When I run the following command to install flight lib  on ubunut20.04 with python3.7 or 3.6:
~/flightmare/flightlib$ python3.7 -m pip install 

I have this error:
Building wheels for collected packages: flightgym
  Building wheel for flightgym (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-pk7qus6b/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-pk7qus6b/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-5nxqmaz3
       cwd: /tmp/pip-req-build-pk7qus6b/
  Complete output (127 lines):
  running bdist_wheel
  running build
  running build_ext
  -- The C compiler identification is GNU 9.3.0
  -- The CXX compiler identification is GNU 9.3.0
  -- Check for working C compiler: /usr/bin/cc
  -- Check for working C compiler: /usr/bin/cc -- works
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Check for working CXX compiler: /usr/bin/c++
  -- Check for working CXX compiler: /usr/bin/c++ -- works
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- ======================  !Flightmare!  ======================
  -- ======> Setup Dependencies
  -- Using system provided Eigen.
  /usr/include/eigen3
  -- Eigen3 include dir: /usr/include/eigen3
  -- Getting Pybind11...
  Cloning into 'pybind11-src'...
  Already on 'master'
  -- Pybind11 downloaded!
  -- pybind11 v2.6.3 dev1
  -- Found PythonInterp: /usr/bin/python3.7 (found version "3.7.9")
  -- Found PythonLibs: python3.7m
  -- Performing Test HAS_FLTO
  -- Performing Test HAS_FLTO - Success
  -- Getting yaml-cpp...
  -- Yaml downloaded!
  -- Found OpenCV: /usr (found version "4.2.0")
  -- Found OpenMP_C: -fopenmp (found version "4.5")
  -- Found OpenMP_CXX: -fopenmp (found version "4.5")
  -- Found OpenMP: TRUE (found version "4.5")
  INFOBuild time could be improved with ccache!
  INFO    sudo apt install ccache
  -- ======> Setup Compilation
  -- Enabling fast optimization flags!
  -- The activated CXX RELEASE configuration is:
   -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -DNDEBUG  -fopenmp  -Ofast  -march=native
  -- The activated CXX DEBUG configuration is:
   -DVERSION_INFO=\"0.0.1\" -fPIC -Wall -g -fopenmp
  -- ======>  Setup Build
  -- Getting gtests...
  Cloning into 'googletest-src'...
  Already on 'master'
  -- gtests downloaded!
  -- Found Python: /usr/bin/python3.9 (found version "3.9.1") found components: Interpreter
  -- Looking for pthread.h
  -- Looking for pthread.h - found
  -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
  -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
  -- Looking for pthread_create in pthreads
  -- Looking for pthread_create in pthreads - not found
  -- Looking for pthread_create in pthread
  -- Looking for pthread_create in pthread - found
  -- Found Threads: TRUE
[... 183 lines cut ...]
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.7/distutils/command/install.py", line 589, in run
        self.run_command('build')
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3.7/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-req-build-pk7qus6b/setup.py", line 59, in run
        self.build_extension(ext)
      File "/tmp/pip-req-build-pk7qus6b/setup.py", line 89, in build_extension
        cmake_args, cwd=self.build_temp, env=env)
      File "/usr/lib/python3.7/subprocess.py", line 363, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-req-build-pk7qus6b', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-req-build-pk7qus6b/build/lib.linux-x86_64-3.7/', '-DPYTHON_EXECUTABLE=/usr/bin/python3.7', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-pk7qus6b/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-pk7qus6b/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-e7885jcg/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/hasan/.local/include/python3.7m/flightgym Check the logs for full command output.

HasanFarag, 2021-02-06:
I did the following:
1. cd flightmare/flightlib/build && cmake .. && make -j10
2. ./test_lib
Output is
Running main() from /home/hasan/Desktop/flightmare/flightlib/externals/googletest-src/googletest/src/gtest_main.cc
[==========] Running 28 tests from 8 test suites.
[----------] Global test environment set-up.
[----------] 5 tests from EigenChecks
[ RUN      ] EigenChecks.EigenVersionOutput
Eigen Version: 3.3.7
[       OK ] EigenChecks.EigenVersionOutput (0 ms)
[ RUN      ] EigenChecks.EigenQuaternionSequence
[       OK ] EigenChecks.EigenQuaternionSequence (0 ms)
[ RUN      ] EigenChecks.EigenQuaternionRotationDirection
[       OK ] EigenChecks.EigenQuaternionRotationDirection (0 ms)
[ RUN      ] EigenChecks.QuaternionCrossMatrix
[       OK ] EigenChecks.QuaternionCrossMatrix (0 ms)
[ RUN      ] EigenChecks.MatrixColumnwiseDotProduct
[       OK ] EigenChecks.MatrixColumnwiseDotProduct (0 ms)
[----------] 5 tests from EigenChecks (1 ms total)

[----------] 4 tests from Integrators
[ RUN      ] Integrators.ManualEulerAccelerationCheck
[       OK ] Integrators.ManualEulerAccelerationCheck (0 ms)
[ RUN      ] Integrators.ManualRungeKuttaAccelerationCheck
[       OK ] Integrators.ManualRungeKuttaAccelerationCheck (0 ms)
[ RUN      ] Integrators.QuadStateInterface
[       OK ] Integrators.QuadStateInterface (1 ms)
[ RUN      ] Integrators.CheckEulerAgainstRungeKutta
[       OK ] Integrators.CheckEulerAgainstRungeKutta (3 ms)
[----------] 4 tests from Integrators (4 ms total)

[----------] 2 tests from Logger
[ RUN      ] Logger.SimpleLogging
[Test]         This is a text stream log.
[Test]         This is an info log.
[Test]         This could be a warning, but just for demo.
[Test]         This could be an error, but just for demo.
[Test]         You can print strings like "text", and formatted numbers like 3.142.
[Test]         You can use the stream operator '<<' just like with 'std::cout'.
[Test]         This can be helpul for printing complex objects like Eigen vector and matrices:
A vector:
0 1 2 3
A Matrix:
1 0 0
0 1 0
0 0 1
[Test]         And also our own defined objects, like so:
A timer:
[Printing] Timing Timer in 1 calls
[Printing] mean|std:  0.0857 | 0 ms    [min|max:  0.0857 | 0.0857 ms]

A QuadState
State at 0s: [0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]
[       OK ] Logger.SimpleLogging (0 ms)
[ RUN      ] Logger.NoColorLogging
[Test]         This is a text stream log.
[Test]         Info:    This is an info log.
[Test]         Warning: This could be a warning, but just for demo.
[Test]         Error:   This could be an error, but just for demo.
[       OK ] Logger.NoColorLogging (0 ms)
[----------] 2 tests from Logger (0 ms total)

[----------] 3 tests from QuadState
[ RUN      ] QuadState.Constructor
[       OK ] QuadState.Constructor (0 ms)
[ RUN      ] QuadState.Accessors
[       OK ] QuadState.Accessors (0 ms)
[ RUN      ] QuadState.Compare
[       OK ] QuadState.Compare (0 ms)
[----------] 3 tests from QuadState (0 ms total)

[----------] 4 tests from QuadrotorDynamics
[ RUN      ] QuadrotorDynamics.Constructor
Quadrotor Dynamics:
mass =             [1.2]
arm_l =            [0.25]
t_BM =             [ 0.177 -0.177 -0.177  0.177]
                   [-0.177 -0.177  0.177  0.177]
                   [0 0 0 0]
inertia =          [0.0281      0      0]
                   [     0 0.0281      0]
                   [     0      0 0.0437]
motor_omega_min =  [150]
motor_omega_max =  [2e+03]
motor_tau_inv =    [20]
thrust_map =       [1.33e-06  0.00384    -1.77]
kappa =            [0.016]
thrust_min =       [0]
thrust_max =       [11.2]
omega_max =        [6 6 6]

     1      1      1      1
 0.177 -0.177 -0.177  0.177
-0.177 -0.177  0.177  0.177
 0.016 -0.016  0.016 -0.016
[       OK ] QuadrotorDynamics.Constructor (0 ms)
[ RUN      ] QuadrotorDynamics.Dynamics
[       OK ] QuadrotorDynamics.Dynamics (0 ms)
[ RUN      ] QuadrotorDynamics.VectorReference
[       OK ] QuadrotorDynamics.VectorReference (0 ms)
[ RUN      ] QuadrotorDynamics.LoadParams
Quadrotor Dynamics:
mass =             [0.73]
arm_l =            [0.17]
t_BM =             [ 0.12 -0.12 -0.12  0.12]
                   [-0.12 -0.12  0.12  0.12]
                   [0 0 0 0]
inertia =          [0.00791       0       0]
                   [      0 0.00791       0]
                   [     0      0 0.0123]
motor_omega_min =  [150]
motor_omega_max =  [3e+03]
motor_tau_inv =    [1e+04]
thrust_map =       [1.33e-06  0.00384    -1.77]
kappa =            [0.016]
thrust_min =       [0]
thrust_max =       [11.2]
omega_max =        [6 6 6]

[       OK ] QuadrotorDynamics.LoadParams (18 ms)
[----------] 4 tests from QuadrotorDynamics (18 ms total)

[----------] 3 tests from QuadrotorEnv
[ RUN      ] QuadrotorEnv.Constructor
[QuadrotorEnv] Environment configuration path 
"/home/hasan/Desktop/flightmare/flightlib/configs/quadrotor_env.yaml".
free(): invalid size
Segmentation fault (core dumped)

Side Note: I tested on ubuntu20.04 docker with venv and still can not manage to install flightlib

yun-long, 2021-02-07:
hi, could you post which gcc and g++ version you are using?
you can check it via
`gcc --version`
or
'g++ --version'

yun-long, 2021-02-07:
There was a problem with gcc/g++ 9.3.0. I updated the master branch, could you pull and rebuild the project again. 
Let me know if the problem still exists.

HasanFarag, 2021-02-08:
yes, i have  gcc/g++ 9.3.0. The cpp tests for flightlib passed. but the same error exists with pybind when installing the python package. It should work on ubuntu 20, right? or ubuntu 18.04 is only tested? i suppose if it works on ubuntu 18.04, should not some much changes to be made to run on 20.04 (ofc, it depend on your dependencies like zeromq,etc).

yun-long, 2021-02-10:
there might be problems if you do both "came .. && make" in the build folder and then do "pip install ." via the setup.py

what you can do is 

- remove everything in flightlib/build and flightlib/external and then `pip install . ` under flightlib/ via[ this setup.py file](https://github.com/uzh-rpg/flightmare/blob/dev/docs/flightlib/setup.py)
- or first build the project in flightlib/build and do pip install . under flightlib/build via 
- 
```
from setuptools import setup, Extension
#
setup(name='flightgym',
      version='0.0.1',
      author="Yunlong Song",
      author_email='song@ifi.uzh.ch',
      description="Flightmare: A Quadrotor Simulator",
      long_description='',
      packages=[''],
      package_dir={'': './'},
      package_data={'': ['flightgym.cpython-36m-x86_64-linux-gnu.so']},
      zip_fase=True,
      url=None,
      )
 
```

closed 2021-02-10 by HasanFarag

driedler, 2021-12-05:
`sudo apt-get install python3.7-dev `

fixed this error for me.

wangziqiaorobot, 2022-02-24:
fixed this error for me too

samaila53, 2023-08-02:
Hi, after following the flightmare install steps , i get this error:

Building wheels for collected packages: flightgym
Building wheel for flightgym (setup.py) ... error
ERROR: Command errored out with exit status 1:
command: /home/sam/anaconda3/envs/quad/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/sam/Bureau/flightmare/flightlib/setup.py'"'"'; file='"'"'/home/sam/Bureau/flightmare/flightlib/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(file) if os.path.exists(file) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, file, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-m2jlvmei
cwd: /home/sam/Bureau/flightmare/flightlib/
Complete output (106 lines):
running bdist_wheel
running build
running build_ext
Removing some cache file: /home/sam/Bureau/flightmare/flightlib/build/temp.linux-x86_64-3.6
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ====================== !Flightmare! ======================
-- ======> Setup Dependencies
-- Using system provided Eigen.
/usr/include/eigen3
-- Eigen3 include dir: /usr/include/eigen3
-- Getting Pybind11...
Cloning into 'pybind11-src'...
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
error: 3262 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
Cloning into 'pybind11-src'...
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 1511 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
Cloning into 'pybind11-src'...
Already on 'master'
-- Pybind11 downloaded!
-- pybind11 v2.12.0 dev1
-- Found PythonInterp: /home/sam/anaconda3/envs/quad/bin/python (found suitable version "3.6.13", minimum required is "3.6")
-- Found PythonLibs: /home/sam/anaconda3/envs/quad/lib/libpython3.6m.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Getting yaml-cpp...
-- Yaml downloaded!
-- Found OpenCV: /usr (found version "4.5.4")
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
INFOBuild time could be improved with ccache!
INFO sudo apt install ccache
-- ======> Setup Compilation
-- Enabling fast optimization flags!
-- The activated CXX RELEASE configuration is:
-DVERSION_INFO="0.0.1" -fPIC -Wall -DNDEBUG -fopenmp -Ofast -march=native
-- The activated CXX DEBUG configuration is:
-DVERSION_INFO="0.0.1" -fPIC -Wall -g -fopenmp
-- ======> Setup Build
-- Getting gtests...
Cloning into 'googletest-src'...
fatal: invalid reference: master
CMake Error at googletest-download/googletest-prefix/tmp/googletest-gitclone.cmake:40 (message):
Failed to checkout tag: 'master'
[... 20 lines cut ...]
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/dist.py", line 974, in run_command
cmd_obj.run()
File "/home/sam/anaconda3/envs/quad/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 299, in run
self.run_command('build')
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/cmd.py", line 313, in run_command
self.distribution.run_command(command)
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/dist.py", line 974, in run_command
cmd_obj.run()
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/command/build.py", line 135, in run
self.run_command(cmd_name)
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/cmd.py", line 313, in run_command
self.distribution.run_command(command)
File "/home/sam/anaconda3/envs/quad/lib/python3.6/distutils/dist.py", line 974, in run_command
cmd_obj.run()
File "/home/sam/Bureau/flightmare/flightlib/setup.py", line 59, in run
self.build_extension(ext)
File "/home/sam/Bureau/flightmare/flightlib/setup.py", line 89, in build_extension
cmake_args, cwd=self.build_temp, env=env)
File "/home/sam/anaconda3/envs/quad/lib/python3.6/subprocess.py", line 311, in check_call
raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '/home/sam/Bureau/flightmare/flightlib', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/sam/Bureau/flightmare/flightlib/build/lib.linux-x86_64-3.6/', '-DPYTHON_EXECUTABLE=/home/sam/anaconda3/envs/quad/bin/python', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.

ERROR: Failed building wheel for flightgym

Ho do solve it?

fahnub, 2023-10-22:
> `sudo apt-get install python3.7-dev `
> 
> fixed this error for me.

Worked for me.