uzh-rpg/flightmare #85: Trying to install flightmare and run RL examples

hhlei, 2021-03-08:
when running python3 run_drone_control.py I get a 'no module named 'rpg_baselines.common'' error. I checked my conda environments however, and it says that rpg-baselines has been installed.

yun-long, 2021-03-15:
this shouldn't happen if you install rpg_baselines correctly. make sure you are using the right python interpreter, e..g, check` which python` and `which python3`, see if it matches the pip you used for installing the rpg_baselines.

phxhuy, 2021-03-22:
Hi, I have also got the same problem. If you have installed  with the same pip and python, and the problem still persist, try to go into the flightrl folder (where there is a setup.py file) and type: pip install -e .
Good luck

HuaYuXiao, 2024-12-06:
[image]

I cannot find module named `rpg_baselines` either, but i did install this package, with output of `pip show` demonstrating it.