uzh-rpg/flightmare #90: Problems running the test_vec_env.py

mslavescu, 2021-03-16:
When trying to run this command:

cd $FLIGHTMARE_PATH/flightrl/examples && python test_vec_env.py

From:
https://github.com/uzh-rpg/flightmare/blob/ddc_challenge/README.md

I get these error:

(flightmare) root@594cca055373:~/challenge/flightmare/flightrl/examples# python test_vec_env.py
Traceback (most recent call last):
  File "test_vec_env.py", line 11, in <module>
    import torch
ModuleNotFoundError: No module named 'torch'

Installed torch to fix it:

pip install torch

Then got this error:

flightmare) root@594cca055373:~/challenge/flightmare/flightrl/examples# python3 test_vec_env.py
Traceback (most recent call last):
  File "test_vec_env.py", line 13, in <module>
    from rpg_baselines.envs import vec_env_wrapper as wrapper
  File "/root/miniconda3/envs/flightmare/lib/python3.6/site-packages/rpg_baselines/envs/vec_env_wrapper.py", line 3, in <module>
    from stable_baselines3.common.vec_env import VecEnv
ModuleNotFoundError: No module named 'stable_baselines3'

Install stable_baselines3 to fix it:

pip install stable_baselines3

Then the script worked and connected to the simulator.

The simulator goes in some kind of minimized mode once the Python script is connected, take a look at that issue.

mslavescu, 2021-03-16:
The simulator window becomes very small after the script is connected:
[image]

lorenzoferrini, 2021-03-17:
I updated the requirements for flightgym package to include stable-baselines3, installing stable baselines should install also pytorch as a requirement. 
For the simulator window we'll look into that but for the moment you can manually increase the window size by dragging the window corner.