uzh-rpg/flightmare #91: ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'tensorflow-gpu'

Shihab-Haque, 2021-03-18:
I was trying to install the dependencies for flight-mare, but when I typed the command: 
pip install -r tensorflow-gpu

it gave the error message:
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'tensorflow-gpu'

same thing happened with the command:
pip install -r scikit-build

yun-long, 2021-03-19:
Hi, sorry. There is a mistake in the instruction. 
You should do 
`pip install tensorflow-gpu`
and 
`pip install scikit-build`

Shihab-Haque, 2021-03-19:
Hi ! Thanks for the response. I also tried that, but when I do, it installs tensorflow-gpu-2.4.1  , which doesn't have tf.contrib 
So I'm getting a new error when trying to run "run_drone_control.py" . It says :  import tensorflow.contrib.layers as tf_layers
ModuleNotFoundError: No module named 'tensorflow.contrib'

Here's the full error message.

`python3 run_drone_control.py --train 0 --render 1
[UnityBridge]  Initializing ZMQ connection!
[UnityBridge]  Initializing ZMQ connections done!
[VecEnv]       Flightmare Bridge is created.
Observations:  15
Actions:  4
image shape: (128, 128)
/root/miniconda3/envs/DRONE/lib/python3.6/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
[UnityBridge]  Trying to Connect Unity.
[...............[UnityBridge]  Flightmare Unity is connected.
env.num_envs :  1
density set to  [1.]
/root/miniconda3/envs/DRONE/lib/python3.6/site-packages/cloudpickle/cloudpickle.py:704: UserWarning: A pickle file created using an old (<=1.4.1) version of cloudpicke is currently being loaded. This is not supported by cloudpickle and will break in cloudpickle 1.7
  "will break in cloudpickle 1.7", category=UserWarning
2021-03-19 13:17:22.316439: W tensorflow/stream_executor/platform/default/dso_loader.cc:60] Could not load dynamic library 'libcudart.so.11.0'; dlerror: libcudart.so.11.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /usr/local/nvidia/lib:/usr/local/nvidia/lib64
2021-03-19 13:17:22.316469: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
Traceback (most recent call last):
  File "run_drone_control.py", line 116, in <module>
    main()
  File "run_drone_control.py", line 111, in main
    model = PPO.load(args.weight)
  File "/root/miniconda3/envs/DRONE/lib/python3.6/site-packages/stable_baselines3/common/base_class.py", line 608, in load
    data, params, pytorch_variables = load_from_zip_file(path, device=device, custom_objects=custom_objects)
  File "/root/miniconda3/envs/DRONE/lib/python3.6/site-packages/stable_baselines3/common/save_util.py", line 402, in load_from_zip_file
    data = json_to_data(json_data, custom_objects=custom_objects)
  File "/root/miniconda3/envs/DRONE/lib/python3.6/site-packages/stable_baselines3/common/save_util.py", line 164, in json_to_data
    deserialized_object = cloudpickle.loads(base64_object)
  File "/root/miniconda3/envs/DRONE/lib/python3.6/site-packages/rpg_baselines/common/policies.py", line 9, in <module>
    from stable_baselines.common.tf_util import batch_to_seq, seq_to_batch
  File "/root/miniconda3/envs/DRONE/lib/python3.6/site-packages/stable_baselines/__init__.py", line 6, in <module>
    from stable_baselines.deepq import DQN
  File "/root/miniconda3/envs/DRONE/lib/python3.6/site-packages/stable_baselines/deepq/__init__.py", line 1, in <module>
    from stable_baselines.deepq.policies import MlpPolicy, CnnPolicy, LnMlpPolicy, LnCnnPolicy
  File "/root/miniconda3/envs/DRONE/lib/python3.6/site-packages/stable_baselines/deepq/policies.py", line 2, in <module>
    import tensorflow.contrib.layers as tf_layers
ModuleNotFoundError: No module named 'tensorflow.contrib'
`

ouafi98, 2021-03-20:
install a version of tensorflow lower than 1.18.

for example:
**pip install tensorflow-gpu==1.15
pip install tensorflow==1.15**