uzh-rpg/flightmare #93: Example not running - ModuleNotFound

EceChaik, 2021-03-19:
Hi everyone,

I just went through the installation instructions in the wiki, and after installing everything, I run the proposed example
python3 run_drone_control.py --train 0

But I get the following error:
python3 run_drone_control.py --train 0
2021-03-19 14:14:08.561713: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.11.0
Traceback (most recent call last):
  File "run_drone_control.py", line 12, in <module>
    from stable_baselines import logger
  File "/home/dc4204/anaconda3/envs/Flightmare/lib/python3.6/site-packages/stable_baselines/__init__.py", line 6, in <module>
    from stable_baselines.deepq import DQN
  File "/home/dc4204/anaconda3/envs/Flightmare/lib/python3.6/site-packages/stable_baselines/deepq/__init__.py", line 1, in <module>
    from stable_baselines.deepq.policies import MlpPolicy, CnnPolicy, LnMlpPolicy, LnCnnPolicy
  File "/home/dc4204/anaconda3/envs/Flightmare/lib/python3.6/site-packages/stable_baselines/deepq/policies.py", line 2, in <module>
    import tensorflow.contrib.layers as tf_layers
ModuleNotFoundError: No module named 'tensorflow.contrib'

Any help appreciated, thanks

yun-long, 2021-03-19:
make sure you install the right version of tensorflow, which should be tensorlfow  <1.18

EceChaik, 2021-03-19:
Hi, thanks for the swift reply.

So what I did now was 

pip uninstall tensorflow
pip install tensorflow==1.15
then again I run
cd flightlib ; pip install .
cd flightrl ; pip install .
cd examples ; python3 run_drone_control.py --train 0

This time the output was:
WARNING:tensorflow:
The TensorFlow contrib module will not be included in TensorFlow 2.0.
For more information, please see:
  * https://github.com/tensorflow/community/blob/master/rfcs/20180907-contrib-sunset.md
  * https://github.com/tensorflow/addons
  * https://github.com/tensorflow/io (for I/O related ops)
If you depend on functionality not listed there, please file an issue.

Traceback (most recent call last):
  File "run_drone_control.py", line 15, in <module>
    from rpg_baselines.common.policies import MlpPolicy
ModuleNotFoundError: No module named 'rpg_baselines.common'

ouafi98, 2021-03-20:
I've faced the same issue. 

Just replace **packages=['rpg_baselines'],** in the file _path_to_flightmare/flightrl/setup.py_ by **packages=['rpg_baselines', 'rpg_baselines.ppo', 'rpg_baselines.common', 'rpg_baselines.envs'],** fixed it for me.

I hope this will help you.
Good luck.

EceChaik, 2021-03-21:
Thanks a lot, it fixed it for me as well.

kuldeepbrd1, 2021-03-25:
In general you can also use `-m` flag in the parent dir, to account for all modules in that dir.

So in `flightrl` dir, running the following also solves the issue. 
`python -m examples.run_drone_control`

samaila53, 2023-08-04:
Traceback (most recent call last):
  File "run_drone_control.py", line 21, in <module>
    from flightgym import QuadrotorEnv_v1
ModuleNotFoundError: No module named 'flightgym'

can someone help me please