uzh-rpg/flightmare #94: ddc_challenge :

jhurlbut, 2021-03-21:
When I try to run the training example run_drone_control.py I get an error: 
Could not load dynamic library 'libcudart.so.11.0'

My host has Cuda 11 installed but the docker image seems to have libraries only for 10.2 in:

/usr/local/cuda-10.2/targets/x86_64-linux/lib/libcudart.so.10.2.89
/usr/local/cuda-10.2/targets/x86_64-linux/lib/libcudart.so.10.2

How can I get the docker image to have the Cuda 11 libraries installed? thank you

arsimone, 2021-03-21:
The image is created building on top of the official image from Nvidia with Cuda 10.2. This issue is quite strange, are you using the scripts from within the docker container?
To solve this you might attempt a manual installation of Cuda 11 in the container, which I do not recommend, or install Cuda 10.2 on your host.

jhurlbut, 2021-03-22:
I was able to solve the issue by downgrading the host to cuda 10.2. I then had to delete and repull the docker image for some reason. Thank you!

closed 2021-03-22 by jhurlbut

jhurlbut, 2021-03-23:
for anyone with Cuda 11 issues, I also had to set the cudatoolkit flag when installing torchvision/pytorch like so:
conda install pytorch==1.7.1 torchvision==0.8.2 torchaudio==0.7.2 cudatoolkit=10.2 -c pytorch

EceChaik, 2021-03-23:
Are you using Ubuntu 18 or 20?

EceChaik, 2021-03-23:
Did you also do
pip install tensorflow-gpu ?
Because for me the "libcudart.11" issue arises after running the above command to install tensorflow-gpu