uzh-rpg/flightmare #95: Forest Rendering Issue

EceChaik, 2021-03-22:
Hi guys,
I am just getting started with the flightmare package, and I am testing the pre-trained RL examples.
After going through the wiki and also downloading the latest release for the Unity rendering, the forest scene is not rendering correctly.
The example still runs and the drone hovers appropriately, but while all the other scenes show appropriately, this is what the forest environment looks like:
[image]

Is this a known issue?

yun-long, 2021-03-22:
this rendering is correct. I guess the current drone position is very close to (x, y, z) = (0, 0, 5). 
If you set the drone position to (x, y, z) = (101, 86, 31). You will have significantly different views.

yun-long, 2021-03-22:
To do so, modify the[ flightlib/src/envs/quadrotor_env/quadrotor_env.cpp](https://github.com/uzh-rpg/flightmare/blob/master/flightlib/src/envs/quadrotor_env/quadrotor_env.cpp)

At the moment, the goal state is (0, 0, 5). change it to (101, 86, 31).
Also adapt the world_box. e.g.,  world_box_ << -200, 200, -200, 200, -200, 200;

And change the reset 
    // randomly reset the quadrotor state
    // reset position
    quad_state_.x(QS::POSX) = uniform_dist_(random_gen_) + 101;
    quad_state_.x(QS::POSY) = uniform_dist_(random_gen_) + 86;
    quad_state_.x(QS::POSZ) = uniform_dist_(random_gen_) + 5;

`pip install .`
retrain your policy.
and now the drone will fly to the new goal position. 

Note, the RL policy is trained for hovering at one specific goal position. Every time you change the goal position, you would have to retrain it. Nevertheless, I guess it serves as a good start for more complex tasks.

EceChaik, 2021-03-23:
Thanks, you were right about the position being bad for the forest environment.
I did the proposed changes but after training the drone just keeps falling. Could it be that some RL parameters need to be adjusted as well, for the new position? 
I wouldn't ask for something like this normally, cause it's clearly not a package issue at this point, I just want to make sure I'm following the correct procedure. This is the final output after running run_drone_control.py with the new goal state:
[image]

and these are the resulting flights. 
[image]

yun-long, 2021-03-24:
no worries. Could you share your code here, I mean the quadrotor_env.cpp

EceChaik, 2021-03-24:
This is my modified quadrotor_env.cpp file:

`#include "flightlib/envs/quadrotor_env/quadrotor_env.hpp"

namespace flightlib {

QuadrotorEnv::QuadrotorEnv()
  : QuadrotorEnv(getenv("FLIGHTMARE_PATH") +
                 std::string("/flightlib/configs/quadrotor_env.yaml")) {}

QuadrotorEnv::QuadrotorEnv(const std::string &cfg_path)
  : EnvBase(),
    pos_coeff_(0.0),
    ori_coeff_(0.0),
    lin_vel_coeff_(0.0),
    ang_vel_coeff_(0.0),
    act_coeff_(0.0),
    goal_state_((Vector<quadenv::kNObs>() << 101.0, 86.0, 31.0, 0.0, 0.0, 0.0, 0.0,
                 0.0, 0.0, 0.0, 0.0, 0.0)
                  .finished()) {
  // load configuration file
  YAML::Node cfg_ = YAML::LoadFile(cfg_path);

  quadrotor_ptr_ = std::make_shared<Quadrotor>();
  // update dynamics
  QuadrotorDynamics dynamics;
  dynamics.updateParams(cfg_);
  quadrotor_ptr_->updateDynamics(dynamics);

  // define a bounding box
  world_box_ << -200, 200, -200, 200, -200, 200;
  if (!quadrotor_ptr_->setWorldBox(world_box_)) {
    logger_.error("cannot set wolrd box");
  };

  // define input and output dimension for the environment
  obs_dim_ = quadenv::kNObs;
  act_dim_ = quadenv::kNAct;

  Scalar mass = quadrotor_ptr_->getMass();
  act_mean_ = Vector<quadenv::kNAct>::Ones() * (-mass * Gz) / 4;
  act_std_ = Vector<quadenv::kNAct>::Ones() * (-mass * 2 * Gz) / 4;

  // load parameters
  loadParam(cfg_);
}

QuadrotorEnv::~QuadrotorEnv() {}

bool QuadrotorEnv::reset(Ref<Vector<>> obs, const bool random) {
  quad_state_.setZero();
  quad_act_.setZero();

  if (random) {
    // randomly reset the quadrotor state
    // reset position
    quad_state_.x(QS::POSX) = uniform_dist_(random_gen_) + 101;
    quad_state_.x(QS::POSY) = uniform_dist_(random_gen_) + 86;
    quad_state_.x(QS::POSZ) = uniform_dist_(random_gen_) + 31;
    if (quad_state_.x(QS::POSZ) < -0.0)
      quad_state_.x(QS::POSZ) = -quad_state_.x(QS::POSZ);
    // reset linear velocity
    quad_state_.x(QS::VELX) = uniform_dist_(random_gen_);
    quad_state_.x(QS::VELY) = uniform_dist_(random_gen_);
    quad_state_.x(QS::VELZ) = uniform_dist_(random_gen_);
    // reset orientation
    quad_state_.x(QS::ATTW) = uniform_dist_(random_gen_);
    quad_state_.x(QS::ATTX) = uniform_dist_(random_gen_);
    quad_state_.x(QS::ATTY) = uniform_dist_(random_gen_);
    quad_state_.x(QS::ATTZ) = uniform_dist_(random_gen_);
[... 112 lines cut ...]
  if (!cmd_.valid()) return false;
  *cmd = cmd_;
  return true;
}

void QuadrotorEnv::addObjectsToUnity(std::shared_ptr<UnityBridge> bridge) {
  bridge->addQuadrotor(quadrotor_ptr_);
}

std::ostream &operator<<(std::ostream &os, const QuadrotorEnv &quad_env) {
  os.precision(3);
  os << "Quadrotor Environment:\n"
     << "obs dim =            [" << quad_env.obs_dim_ << "]\n"
     << "act dim =            [" << quad_env.act_dim_ << "]\n"
     << "sim dt =             [" << quad_env.sim_dt_ << "]\n"
     << "max_t =              [" << quad_env.max_t_ << "]\n"
     << "act_mean =           [" << quad_env.act_mean_.transpose() << "]\n"
     << "act_std =            [" << quad_env.act_std_.transpose() << "]\n"
     << "obs_mean =           [" << quad_env.obs_mean_.transpose() << "]\n"
     << "obs_std =            [" << quad_env.obs_std_.transpose() << std::endl;
  os.precision();
  return os;
}

}  // namespace flightlib`

EceChaik, 2021-04-04:
Have you perhaps spotted any mistakes?