uzh-rpg/flightmare #96: Cameras for multiple quadrotors

anoopsonar30, 2021-03-22:
I modified racing.cpp and added multiple quadrotors but the camera only follows one of them. How would I go about keeping the camera static/fixed at some position in the environment? Or have 1 camera for each drone?

anoopsonar30, 2021-03-22:
```
  // unity quadrotors
  std::vector<std::shared_ptr<Quadrotor>> quad_ptr(numDrones);

  for (int i = 0; i < numDrones; i++){
    std::shared_ptr<Quadrotor> quad = *(new std::shared_ptr<Quadrotor>(std::make_shared<Quadrotor>()));
    quad_ptr.push_back(quad);
  }

  // std::shared_ptr<Quadrotor> quad_ptr1 = std::make_shared<Quadrotor>();
  // std::shared_ptr<Quadrotor> quad_ptr2 = std::make_shared<Quadrotor>();
  
```

Also my code crashes when I declare the quadrotors using a for loop as shown above but if I declared separate quadrotor objects outside a loop (which is not a scalable way of writing code) then the programs works just fine. 

```
[drones-3] process has died [pid 625906, exit code -11, cmd /home/snoop/catkin_ws/devel/lib/thesis/drones __name:=drones __log:=/home/snoop/.ros/log/38e57ea0-8b36-11eb-9636-f5a078189b74/drones-3.log].
log file: /home/snoop/.ros/log/38e57ea0-8b36-11eb-9636-f5a078189b74/drones-3*.log
[rpg_flightmare_render-2] process has finished cleanly
log file: /home/snoop/.ros/log/38e57ea0-8b36-11eb-9636-f5a078189b74/rpg_flightmare_render-2*.log
```

anoopsonar30, 2021-03-22:
@yun-long

yun-long, 2021-03-22:
could you share all your code in the racing.cpp file?

anoopsonar30, 2021-03-22:
This is drones.cpp - 
```
// ros
#include <cv_bridge/cv_bridge.h>
#include <image_transport/image_transport.h>
#include <ros/ros.h>

// flightlib
#include "flightlib/bridges/unity_bridge.hpp"
#include "flightlib/bridges/unity_message_types.hpp"
#include "flightlib/common/quad_state.hpp"
#include "flightlib/common/types.hpp"
#include "flightlib/objects/quadrotor.hpp"
#include "flightlib/objects/static_gate.hpp"
#include "flightlib/sensors/rgb_camera.hpp"

// trajectory
#include <polynomial_trajectories/minimum_snap_trajectories.h>
#include <polynomial_trajectories/polynomial_trajectories_common.h>
#include <polynomial_trajectories/polynomial_trajectory.h>
#include <polynomial_trajectories/polynomial_trajectory_settings.h>
#include <quadrotor_common/trajectory_point.h>

// Constants
const double EPSILON = 1e-6;

using namespace flightlib;

bool destinationCheck(quadrotor_common::TrajectoryPoint desired_pose, Eigen::Vector3d pos){
  return fabs((Scalar)desired_pose.position.x() - pos[0]) < EPSILON && 
  fabs((Scalar)desired_pose.position.y() - pos[1]) < EPSILON && 
  fabs((Scalar)desired_pose.position.z() - pos[2]) < EPSILON;
}

int main(int argc, char *argv[]) {
  // initialize ROS
  ros::init(argc, argv, "Multi Drone");
  ros::NodeHandle nh("");
  ros::NodeHandle pnh("~");
  ros::Rate(60.0);

  // Flightmare(Unity3D)
  std::shared_ptr<UnityBridge> unity_bridge_ptr = UnityBridge::getInstance();
  SceneID scene_id{UnityScene::WAREHOUSE};
  bool unity_ready{false};
  int numDrones = 2;

  // unity quadrotors
  std::vector<std::shared_ptr<Quadrotor>> quad_ptr(numDrones);

  for (int i = 0; i < numDrones; i++){
    std::shared_ptr<Quadrotor> quad = *(new std::shared_ptr<Quadrotor>(std::make_shared<Quadrotor>()));
    quad_ptr.push_back(quad);
  }

  // std::shared_ptr<Quadrotor> quad_ptr1 = std::make_shared<Quadrotor>();
  // std::shared_ptr<Quadrotor> quad_ptr2 = std::make_shared<Quadrotor>();
  
  QuadState quad_state1;
  QuadState quad_state2;
  
  quad_state1.setZero();
  quad_state2.setZero();

  quad_ptr[0]->reset(quad_state1);
  quad_ptr[1]->reset(quad_state2);

  // define quadsize scale (for unity visualization only)
  Vector<3> quad_size(0.5, 0.5, 0.5);

[... 86 lines cut ...]
    quad_state2.x[QS::POSX] = (Scalar)desired_pose2.position.x();
    quad_state2.x[QS::POSY] = (Scalar)desired_pose2.position.y();
    quad_state2.x[QS::POSZ] = (Scalar)desired_pose2.position.z();
    quad_state2.x[QS::ATTW] = (Scalar)desired_pose2.orientation.w();
    quad_state2.x[QS::ATTX] = (Scalar)desired_pose2.orientation.x();
    quad_state2.x[QS::ATTY] = (Scalar)desired_pose2.orientation.y();
    quad_state2.x[QS::ATTZ] = (Scalar)desired_pose2.orientation.z();
    quad_ptr[1]->setState(quad_state2);

    unity_bridge_ptr->getRender(frame_id);
    unity_bridge_ptr->handleOutput();

    frame_id += 1;

    // Check if you're sufficiently close to the destination
    // reachedDestination = destinationCheck(desired_pose, end);
    // if (reachedDestination)
    //   break;
  }

  return 0;
}
```

Eventually I would like to have 10+ drones so I will need to use loops.

anoopsonar30, 2021-03-22:
@yun-long

anoopsonar30, 2021-03-23:
@yun-long

yun-long, 2021-03-24:
`
// ros
#include <cv_bridge/cv_bridge.h>
#include <image_transport/image_transport.h>
#include <ros/ros.h>

// flightlib
#include "flightlib/bridges/unity_bridge.hpp"
#include "flightlib/bridges/unity_message_types.hpp"
#include "flightlib/common/quad_state.hpp"
#include "flightlib/common/types.hpp"
#include "flightlib/objects/quadrotor.hpp"
#include "flightlib/objects/static_gate.hpp"
#include "flightlib/sensors/rgb_camera.hpp"

// trajectory
#include <polynomial_trajectories/minimum_snap_trajectories.h>
#include <polynomial_trajectories/polynomial_trajectories_common.h>
#include <polynomial_trajectories/polynomial_trajectory.h>
#include <polynomial_trajectories/polynomial_trajectory_settings.h>
#include <quadrotor_common/trajectory_point.h>

using namespace flightlib;

int main(int argc, char *argv[]) {
  // initialize ROS
  ros::init(argc, argv, "gates_example");
  ros::NodeHandle nh("");
  ros::NodeHandle pnh("~");
  ros::Rate(50.0);

  // Flightmare(Unity3D)
  std::shared_ptr<UnityBridge> unity_bridge_ptr = UnityBridge::getInstance();
  SceneID scene_id{UnityScene::WAREHOUSE};
  bool unity_ready{false};

  // unity quadrotor
  int num_drones = 2 + 1;
  std::vector<std::shared_ptr<Quadrotor>> quad_ptrs;

  for (int i = 0; i < num_drones; i++) {
    std::shared_ptr<Quadrotor> quad = std::make_shared<Quadrotor>();
    QuadState quad_state;
    quad_state.setZero();
    quad->reset(quad_state);
    //
    Vector<3> quad_size(0.5, 0.5, 0.5);
    if (i == 1) {
      quad_size << 0.0000, 0.0000, 0.0000;
    }
    // define quadsize scale (for unity visualization only)
    quad->setSize(quad_size);
    //
    quad_ptrs.push_back(quad);
    unity_bridge_ptr->addQuadrotor(quad);
  }
  std::cout << quad_ptrs.size() << std::endl;

  // Initialize gates
  std::string object_id = "unity_gate";
  std::string prefab_id = "rpg_gate";
  std::shared_ptr<StaticGate> gate_1 =
    std::make_shared<StaticGate>(object_id, prefab_id);
  gate_1->setPosition(Eigen::Vector3f(5, 0, 2.5));
  gate_1->setRotation(
    Quaternion(std::cos(1 * M_PI_2), 0.0, 0.0, std::sin(1 * M_PI_2)));

  std::string object_id_2 = "unity_gate_2";
  std::shared_ptr<StaticGate> gate_2 =
    std::make_shared<StaticGate>(object_id_2, prefab_id);
  gate_2->setPosition(Eigen::Vector3f(-5, 0, 2.5));
  gate_2->setRotation(
    Quaternion(std::cos(1 * M_PI_2), 0.0, 0.0, std::sin(1 * M_PI_2)));

  // Define path through gates
  std::vector<Eigen::Vector3d> way_points;
  way_points.push_back(Eigen::Vector3d(0, 10, 2.5));
  way_points.push_back(Eigen::Vector3d(5, 0, 2.5));
  way_points.push_back(Eigen::Vector3d(0, -10, 2.5));
  way_points.push_back(Eigen::Vector3d(-5, 0, 2.5));

  std::size_t num_waypoints = way_points.size();
  Eigen::VectorXd segment_times(num_waypoints);
  segment_times << 10.0, 10.0, 10.0, 10.0;
  Eigen::VectorXd minimization_weights(num_waypoints);
  minimization_weights << 0.0, 1.0, 1.0, 1.0, 1.0;

  polynomial_trajectories::PolynomialTrajectorySettings trajectory_settings =
    polynomial_trajectories::PolynomialTrajectorySettings(
      way_points, minimization_weights, 7, 4);

  polynomial_trajectories::PolynomialTrajectory trajectory =
    polynomial_trajectories::minimum_snap_trajectories::
      generateMinimumSnapRingTrajectory(segment_times, trajectory_settings,
                                        20.0, 20.0, 6.0);

  // Start racing
  ros::Time t0 = ros::Time::now();

  unity_bridge_ptr->addStaticObject(gate_1);
  unity_bridge_ptr->addStaticObject(gate_2);
  // connect unity
  unity_ready = unity_bridge_ptr->connectUnity(scene_id);

  FrameID frame_id = 0;
  while (ros::ok() && unity_ready) {
    quadrotor_common::TrajectoryPoint desired_pose =
      polynomial_trajectories::getPointFromTrajectory(
        trajectory, ros::Duration((ros::Time::now() - t0)));
    //
    QuadState quad_state;
    quad_state.setZero();
    quad_state.x[QS::POSX] = -10.0;
    quad_state.x[QS::POSY] = 0.0;
    quad_state.x[QS::POSZ] = 5.0;
    quad_ptrs[1]->setState(quad_state);

    quad_state.x[QS::POSX] = (Scalar)desired_pose.position.x();
    quad_state.x[QS::POSY] = (Scalar)desired_pose.position.y();
    quad_state.x[QS::POSZ] = (Scalar)desired_pose.position.z();
    quad_state.x[QS::ATTW] = (Scalar)desired_pose.orientation.w();
    quad_state.x[QS::ATTX] = (Scalar)desired_pose.orientation.x();
    quad_state.x[QS::ATTY] = (Scalar)desired_pose.orientation.y();
    quad_state.x[QS::ATTZ] = (Scalar)desired_pose.orientation.z();
    quad_ptrs[0]->setState(quad_state);

    quad_state.x[QS::POSX] += 1.0;
    quad_ptrs[2]->setState(quad_state);

    unity_bridge_ptr->getRender(frame_id);
    unity_bridge_ptr->handleOutput();

    //
    frame_id += 1;
  }

  return 0;
}`

1) There was a bug in the trajectory generation. Use the code above. you can use vector for multiple quadrotors
2) at the moment, the third-person-view camera tries to follow one quadrotor. For example, if you have 3 quadrotors, it will follow the second quadrotor. A hacky solution is to define one additional camera, as shown above. Suppose you only need 2 quadrotor, but instead, you can define 3 quads. By fixing the position of the second quad, you can have a "fixed camera".

anoopsonar30, 2021-03-29:
Thanks @yun-long 

Your example shows this using the MinimumSnapRingTrajectory function. Is there an example for the normal point to point MinimumSnapTrajectory function? I'm not sure how the segment times and minimization weights should be set exactly.

anoopsonar30, 2021-03-29:
Nevermind it seems the bug was actually because I was using `std::vector<std::shared_ptr> quad_ptrs(numDrones);` instead of `std::vector<std::shared_ptr> quad_ptrs;`

Very strange

anoopsonar30, 2021-03-29:
```
  std::vector<std::vector<Eigen::Vector3d>*> way_points;
  std::vector<Eigen::Vector3d*> starts;
  std::vector<Eigen::Vector3d*> ends;
  std::vector<quadrotor_common::TrajectoryPoint*> startTraj;
  std::vector<quadrotor_common::TrajectoryPoint*> endTraj;
  
  for (int i = 0; i < numDrones; i++)
  {
    starts.push_back(new Eigen::Vector3d(2 * (i + 1), -2 * i, 2.5));
    ends.push_back(new Eigen::Vector3d(-2 * (i + 1), 2 * i, 2.5));

    quadrotor_common::TrajectoryPoint* initPoint = new quadrotor_common::TrajectoryPoint();
    quadrotor_common::TrajectoryPoint* finalPoint = new quadrotor_common::TrajectoryPoint();
    initPoint->position = *starts[i];
    finalPoint->position = *ends[i];
    startTraj.push_back(initPoint);
    endTraj.push_back(finalPoint);

    std::vector<Eigen::Vector3d> *waypoint = new std::vector<Eigen::Vector3d>();
    waypoint->push_back(*starts[i]);
    waypoint->push_back(*ends[i]);
    way_points.push_back(waypoint);
  }

  std::size_t num_waypoints = way_points[0]->size();
  Eigen::VectorXd segment_times(num_waypoints + 1);
  segment_times << 30.0, 30.0, 30.0;
  Eigen::VectorXd minimization_weights(num_waypoints);
  minimization_weights << 1.0, 1.0;

  std::vector<polynomial_trajectories::PolynomialTrajectorySettings> trajSettings;
  std::vector<polynomial_trajectories::PolynomialTrajectory> trajs;

  for (int i = 0; i < numDrones; i++)
  {
    polynomial_trajectories::PolynomialTrajectorySettings setting = 
    polynomial_trajectories::PolynomialTrajectorySettings(
    *way_points[i], minimization_weights, 7, 4);    
    trajSettings.push_back(setting);

    polynomial_trajectories::PolynomialTrajectory traj = polynomial_trajectories::minimum_snap_trajectories::
      generateMinimumSnapTrajectory(segment_times, *startTraj[i], *endTraj[i], trajSettings[i],
      20.0, 20.0, 6.0);
    trajs.push_back(traj);
  }
```

This throws the following runtime error - 

```
free(): double free detected in tcache 2
[drones-3] process has died [pid 317322, exit code -6, cmd /home/snoop/catkin_ws/devel/lib/thesis/drones __name:=drones __log:=/home/snoop/.ros/log/c5309968-903f-11eb-b8c5-d584db60cac7/drones-3.log].
log file: /home/snoop/.ros/log/c5309968-903f-11eb-b8c5-d584db60cac7/drones-3*.log
```

Do you see what's wrong?

anoopsonar30, 2021-03-29:
@yun-long

francofusco, 2021-03-29:
Hi @anoopsonar30, I just saw your code and I see something wrong. Not 100% sure if this is causing your issue, but it might be worth checking it :relaxed:  

Your code:

> // unity quadrotors
>   std::vector<std::shared_ptr<Quadrotor>> quad_ptr(numDrones);
>
>   for (int i = 0; i < numDrones; i++){
>     std::shared_ptr<Quadrotor> quad = *(new [...]

Has 2 issues:
- How you create and insert objects;
- The way you create the objects.

I'll try to explain why and how to solve the problem :+1: 

----

**Construction of the vector**

When you call `std::vector<T> vec(N)` you already insert `N` instances inside the vector. Since the contained type is "pointer-like", you create a vector that initially contains `N` elements, all equal to `nullptr`. This is not necessarily bad, but later in your code you call `push_back`, which *appends* to the end of the vector. You can solve it by either:

- Calling the default constructor of `std::vector` (which creates an empty vector) and using `push_back` to insert elements.
- Keep the construction as in your example, and *replace* elements using something like `vec[i] = new_value;`.

A draft of the two solutions:

```
// METHOD 1: use std::vector's default constructor
std::vector<std::shared_ptr<Quadrotor>> quad_ptr;
for (int i = 0; i < numDrones; i++) {
  std::shared_ptr<Quadrotor> quad = ...; // we will come back to this later
  quad_ptr.push_back(quad);
}

// METHOD 2: use operator[]
std::vector<std::shared_ptr<Quadrotor>> quad_ptr(numDrones);
for (int i = 0; i < numDrones; i++) {
  std::shared_ptr<Quadrotor> quad = ...; // we will come back to this later
  quad_ptr[i] = quad;
}
```

----

**Proper creation of Quadrotor objects**

The second issue is in the way you instanciate Quadrotors. It is done using the so-called *smart pointers* (the `std::shared_ptr` stuff), which are a sort of pointer, with the exception that they manage dynamic (de)allocations for you. I will not go into detail - but I suggest you to look for further info online: when used properly they are incredibly useful in C++ :muscle:

Anyway, in your code you are dynamically creating pointers to `std::shared_ptr`. This is legal, but often not what one wants. In your code, change:

```c++
std::shared_ptr<Quadrotor> quad = std::make_shared<Quadrotor>();
```
into
```c++
std::shared_ptr<Quadrotor> quad = std::make_shared<Quadrotor>();
```

For reference, `std::make_shared<T>()` makes sure to create an object of type `T` using the call `T* obj = new T()` and then wraps this instance inside a `std::shared_ptr` which is then returned to you. The returned smart pointer can then be stored inside the vector with no problem :+1: Also, you now do not have to manage the dynamically-allocated memory: the smart pointers will make sure to call `delete` when appropriate :smile: 

----

**Final note**

I have not checked your code in depth, so this might not be the actual issue. However, keep in mind that `*(new Object())` is never a good idea in C++ :sweat_smile: All instances that you create with `new` should be eventually released by calling `delete` as well. The fun of using smart pointers is that this is done for you behind the scenes :tada:

Also, I have not checked your [last code snippet](https://github.com/uzh-rpg/flightmare/issues/96#issuecomment-809042363), so the issue might of course be there :thinking: 

Anyway, I hope this feedback was useful. Cheers!

anoopsonar30, 2021-03-29:
@francofusco thanks a lot for the clarification. I see what you mean. I did end up fixing the share_ptr snippet and it works. Would you mind taking a look at the more recent code snippet I shared?

anoopsonar30, 2021-03-29:
I tried writing a shared_ptr implementation instead of using new/delete 

```
  std::vector<std::shared_ptr<std::vector<Eigen::Vector3d>>> way_points;
  std::vector<std::shared_ptr<Eigen::Vector3d>> starts;
  std::vector<std::shared_ptr<Eigen::Vector3d>> ends;
  std::vector<std::shared_ptr<quadrotor_common::TrajectoryPoint>> startTraj;
  std::vector<std::shared_ptr<quadrotor_common::TrajectoryPoint>> endTraj;
  
  for (int i = 0; i < numDrones; i++)
  {
    std::shared_ptr<Eigen::Vector3d> start = std::make_shared<Eigen::Vector3d>(2 * (i + 1), -2 * i, 2.5);
    std::shared_ptr<Eigen::Vector3d> end = std::make_shared<Eigen::Vector3d>(2 * (i + 1), -2 * i, 2.5);
    starts.push_back(start);
    ends.push_back(end);

    std::shared_ptr<quadrotor_common::TrajectoryPoint> initPoint = std::make_shared<quadrotor_common::TrajectoryPoint>();
    std::shared_ptr<quadrotor_common::TrajectoryPoint> finalPoint = std::make_shared<quadrotor_common::TrajectoryPoint>();
    initPoint->position = *start;
    finalPoint->position = *end;
    startTraj.push_back(initPoint);
    endTraj.push_back(finalPoint);

    std::shared_ptr<std::vector<Eigen::Vector3d>> waypoint = std::make_shared<std::vector<Eigen::Vector3d>>();
    waypoint->push_back(*start);
    waypoint->push_back(*end);
    way_points.push_back(waypoint);
  }

  std::size_t num_waypoints = way_points[0]->size();
  Eigen::VectorXd segment_times(num_waypoints + 1);
  segment_times << 30.0, 30.0, 30.0;
  Eigen::VectorXd minimization_weights(num_waypoints);
  minimization_weights << 1.0, 1.0;

  std::vector<polynomial_trajectories::PolynomialTrajectorySettings>> trajSettings;
  std::vector<polynomial_trajectories::PolynomialTrajectory>> trajs;

  for (int i = 0; i < numDrones; i++)
  {
    polynomial_trajectories::PolynomialTrajectorySettings setting = 
    polynomial_trajectories::PolynomialTrajectorySettings(
    *way_points[i], minimization_weights, 7, 4);    
    trajSettings.push_back(setting);

    polynomial_trajectories::PolynomialTrajectory traj = polynomial_trajectories::minimum_snap_trajectories::
      generateMinimumSnapTrajectory(segment_times, *startTraj[i], *endTraj[i], trajSettings[i],
      20.0, 20.0, 6.0);
    trajs.push_back(traj);
  }

```

but now it throws - 
```
[ERROR] [1617021364.649132438]: [/drones] Could not solve quadratic program.
[ERROR] [1617021364.649923907]: [/drones] Could not solve quadratic program.
free(): double free detected in tcache 2
```

francofusco, 2021-03-29:
Hi @anoopsonar30, I have tested a version of your code that does not use smart pointers at all and was able to find the potential culprit. The class `PolynomialTrajectory` does not provide an explicit copy-ctor, and the one generated by default by the compiler fails. My (wild) guess is that this is related to Eigen's memory allocation mechanism when using fixed-size vectors. I have added a quick-n-dirty copy-ctor manually to both `TrajectoryPoint` and `PolynomialTrajectory` and got a successful run which is why my guess is that it is related to copy constructors. Unfortunately, I have started using flightmare only last Thursday and thus I am not familiar enough with the library to be able to say if this is actually the case :sweat_smile:

anoopsonar30, 2021-03-29:
@francofusco I agree, it seemed to me that it had something to do with copy c-tor as well. Would you mind sharing your modified code with me, whether it's some fork of the repo or just paste the copy c-tors here?

francofusco, 2021-03-29:
Sorry but for the moment I cannot :sweat: I already switched off the pc and went back home (I'm replying from my phone) and tomorrow I'll be traveling... I can post it on Wednesday however :relaxed: if you're in a hurry, here's a rough description of my solution. What I did was simply to implement explicitly the copy constructors of both mentioned classes (`TrajectoryPoint` and `PolynomialTrajectory`) so that they in turn copy-construct each of their fields. Something like this:

```c++
TrajectoryPoint::TrajectoryPoint(const TrajectoryPoint& other)
: var1(other.var1)
, var2(other.var2)
// etc for every field!
{ }
```

Not sure if this is unnecessary, but it worked :+1:

anoopsonar30, 2021-04-01:
@francofusco Could you post it once you get the chance? Thanks

anoopsonar30, 2021-04-02:
@francofusco @yun-long It worked! Thank you so much. You guys are legends.

anoopsonar30, 2021-04-02:
@yun-long btw it's not the 1st drone that the camera is fixed on but the (numDrones - 1)th drone which has index numDrones - 2.

closed 2021-04-18 by anoopsonar30

randombot42, 2023-01-19:
Hi @anoopsonar30, I'm also interested in trying multiple drones out for my project. Can you please share the code if you got it working?