uzh-rpg/flightmare #97: Installing -Queries

Roopesh-Bharatwaj-K-R, 2021-03-22:
Packages Flight mare requires CMake and GCC compiler. You will also need system packages python3, OpenMPI, and OpenCV.

apt-get update && apt-get install -y --no-install-recommends \
build-essential \
cmake \
libzmqpp-dev \
libopencv-dev

facing issues here ,  where exactly i should install this packages

yun-long, 2021-03-24:
These are ubuntu packages. 
You should open the ubuntu terminal, for example, use ctrl + alt + T

and then type 

`sudo apt-get update && apt-get install -y --no-install-recommends
build-essential
cmake
libzmqpp-dev
libopencv-dev`

Roopesh-Bharatwaj-K-R, 2021-03-24:
Thank you very much, as I'm using Windows can i also be able to do ? or i should install Ubuntu

yun-long, 2021-03-24:
It is highly recommended to use Ubuntu. All code has been tested only in Ubuntu and we don't have a plan to support Windows. Sorry for the inconvenience.

Roopesh-Bharatwaj-K-R, 2021-03-24:
Thank you very much 
, I will Install and will be back