uzh-rpg/flightmare #103: OMPL for Motion Planning

anoopsonar30, 2021-03-25:
Hi! I installed Open3D, OMPL and OpenMP but am still running into this error while building flightros with motion planning using OMPL.

```
Errors     << flightros:make /home/snoop/catkin_ws/logs/flightros/build.make.063.log                                    
/home/snoop/catkin_ws/src/flightmare/flightros/src/motion_planning/motion_planning.cpp: In function ‘std::vector<motion_planning::float3> motion_planning::readPointCloud()’:
/home/snoop/catkin_ws/src/flightmare/flightros/src/motion_planning/motion_planning.cpp:24:17: warning: unused variable ‘size_mb’ [-Wunused-variable]
   24 |     const float size_mb = file_stream->tellg() * float(1e-6);
      |                 ^~~~~~~
/usr/bin/ld: CMakeFiles/motion_planning.dir/src/motion_planning/motion_planning.cpp.o: in function `ompl::base::SE3StateSpace::SE3StateSpace()':
motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x3d): undefined reference to `ompl::base::StateSpace::getName() const'
/usr/bin/ld: motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x5f): undefined reference to `ompl::base::StateSpace::setName(std::string const&)'
/usr/bin/ld: motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x1a2): undefined reference to `ompl::base::StateSpace::getName() const'
/usr/bin/ld: motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x1bf): undefined reference to `ompl::base::StateSpace::setName(std::string const&)'
/usr/bin/ld: motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x2a3): undefined reference to `ompl::base::StateSpace::getName() const'
/usr/bin/ld: motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x2c0): undefined reference to `ompl::base::StateSpace::setName(std::string const&)'
/usr/bin/ld: CMakeFiles/motion_planning.dir/src/motion_planning/motion_planning.cpp.o: in function `main':
motion_planning.cpp:(.text.startup+0xbdc): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
/usr/bin/ld: motion_planning.cpp:(.text.startup+0xc5d): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
/usr/bin/ld: motion_planning.cpp:(.text.startup+0xccf): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/motion_planning.dir/build.make:373: /home/snoop/catkin_ws/devel/lib/flightros/motion_planning] Error 1
make[1]: *** [CMakeFiles/Makefile2:994: CMakeFiles/motion_planning.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
cd /home/snoop/catkin_ws/build/flightros; catkin build --get-env flightros | catkin env -si  /usr/bin/make --jobserver-auth=4,5; cd -
........................................................................................................................
Failed     << flightros:make                                [ Exited with code 2 ]                                      
Failed    <<< flightros                                     [ 8.0 seconds ]                                             
[build] Summary: 33 of 34 packages succeeded.                                                                           
[build]   Ignored:   2 packages were skipped or are blacklisted.                                                        
[build]   Warnings:  None.                                                                                              
[build]   Abandoned: None.                                                                                              
[build]   Failed:    1 packages failed.    
```     

It seems like a linking error and I've tried editing the CMakeLists.txt but it still fails with the same errors. The linker is failing not just on ompl functions but also on the ros functions so I'm not sure what the problem is.

anoopsonar30, 2021-03-25:
@yun-long

yun-long, 2021-03-29:
this is a linker error. the motion_planning target is not linked to the ROS library properly. 
Not sure why this error appears on your machine. 

https://github.com/uzh-rpg/flightmare/blob/master/flightros/CMakeLists.txt#L103-L112
could you do message(${catkin_LIBRARIES}})
and see what are the outputs when you run `catkin build`

t-thanh, 2021-04-07:
@yun-long : I have the same error as @anoopsonar30 mentioned. I have ROS Noetic on Ubuntu 20.04, have no issue until I tried to compile the motion planning code.

Output of my `message(${catkin_LIBRARIES}})` is bellow (can't find libompl there) but CMake detect the correct OMPL.
```
Found OpenMP TRUE     OpenMP::OpenMP_CXX
Found Open3D 0.12.0
Found OMPL 1.5.2
/home/unix/ros/noetic_ws/devel/.private/flightlib/lib/libflightlib.so/home/unix/ros/noetic_ws/devel/.private/autopilot/lib/libautopilot_helper.so/home/unix/ros/noetic_ws/devel/.private/position_controller/lib/libposition_controller.so/home/unix/ros/noetic_ws/devel/.private/state_predictor/lib/libstate_predictor.so/home/unix/ros/noetic_ws/devel/.private/trajectory_generation_helper/lib/libtrajectory_generation_helper.so/home/unix/ros/noetic_ws/devel/.private/polynomial_trajectories/lib/libpolynomial_trajectories.so/home/unix/ros/noetic_ws/devel/.private/quadrotor_common/lib/libquadrotor_common.so/opt/ros/noetic/lib/libtf_conversions.so/opt/ros/noetic/lib/libkdl_conversions.so/opt/ros/noetic/lib/libtf.so/opt/ros/noetic/lib/libtf2_ros.so/opt/ros/noetic/lib/libactionlib.so/opt/ros/noetic/lib/libtf2.so/opt/ros/noetic/lib/libeigen_conversions.so/usr/lib/liborocos-kdl.so/opt/ros/noetic/lib/libimage_transport.so/opt/ros/noetic/lib/libmessage_filters.so/opt/ros/noetic/lib/libclass_loader.so/usr/lib/x86_64-linux-gnu/libPocoFoundation.so/usr/lib/x86_64-linux-gnu/libdl.so/opt/ros/noetic/lib/libroscpp.so/usr/lib/x86_64-linux-gnu/libpthread.so/usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.71.0/opt/ros/noetic/lib/libxmlrpcpp.so/opt/ros/noetic/lib/libroslib.so/opt/ros/noetic/lib/librospack.so/usr/lib/x86_64-linux-gnu/libpython3.8.so/usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0/usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.71.0/usr/lib/x86_64-linux-gnu/libtinyxml2.so/opt/ros/noetic/lib/libcv_bridge.so/usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_dnn.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_features2d.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_flann.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_ml.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_photo.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_stitching.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_video.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_videoio.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_aruco.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_bgsegm.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_bioinspired.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_ccalib.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_datasets.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_dnn_objdetect.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_dnn_superres.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_dpm.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_face.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_freetype.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_fuzzy.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_hdf.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_hfs.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_img_hash.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_line_descriptor.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_optflow.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_phase_unwrapping.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_plot.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_quality.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_reg.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_rgbd.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_saliency.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_shape.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_stereo.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_structured_light.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_superres.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_surface_matching.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_text.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_tracking.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_videostab.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_viz.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_ximgproc.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_xobjdetect.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_xphoto.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_core.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2.0/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.4.2.0/opt/ros/noetic/lib/librosconsole.so/opt/ros/noetic/lib/librosconsole_log4cxx.so/opt/ros/noetic/lib/librosconsole_backend_interface.so/usr/lib/x86_64-linux-gnu/liblog4cxx.so/usr/lib/x86_64-linux-gnu/libboost_regex.so.1.71.0/opt/ros/noetic/lib/libroscpp_serialization.so/opt/ros/noetic/lib/librostime.so/usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.71.0/opt/ros/noetic/lib/libcpp_common.so/usr/lib/x86_64-linux-gnu/libboost_system.so.1.71.0/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0/usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4}}
```

silvery107, 2021-08-17:
**@yun-long**: Hi Yunlong, I have the similar error as **@anoopsonar30** mentioned. I failed to compile the `motion_planning` code but with undefined reference to multiple functions in `ompl::base::StateSpace`. 
It seems not like a linking error to the ROS library. Is it result from version conflicts of external packages (i.e. OMPL, OpenMP)?

```shell
Found OpenMP TRUE   4  OpenMP::OpenMP_CXX
Found Open3D 0.13.0
Found OMPL 1.4.2
CMakeFiles/motion_planning.dir/src/motion_planning/motion_planning.cpp.o：in function 'ompl::base::SE3StateSpace::SE3StateSpace()'：
motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x39)：undefined reference to 'ompl::base::StateSpace::getName() const'
motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x5b)：undefined reference to 'ompl::base::StateSpace::setName(std::string const&)'
motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x19d)：undefined reference to 'ompl::base::StateSpace::getName() const'
motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x1ba)：undefined reference to 'ompl::base::StateSpace::setName(std::string const&)'
motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x2d9)：undefined reference to 'ompl::base::StateSpace::getName() const'
motion_planning.cpp:(.text._ZN4ompl4base13SE3StateSpaceC2Ev[_ZN4ompl4base13SE3StateSpaceC5Ev]+0x2f6)：undefined reference to 'ompl::base::StateSpace::setName(std::string const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/silvery/Courses/drone_ws/devel/lib/flightros/motion_planning] Error 1
make[1]: *** [CMakeFiles/motion_planning.dir/all] Error 2
make: *** [all] Error 2
```

yhy-lk, 2023-05-05:
@silvery107 :hi,I have the same problem as yours, did you solve it?