uzh-rpg/flightmare #104: Get RBG Image Observation

oggyfaker, 2021-03-26:
Hi authors, 

i have some issues with getting RGB images. I found  bool QuadrotorEnv::getImages(Ref<DepthImage<>> img) in quadrotor_env.cpp has rgb_camera->getDepthMap(img_depth_), it has  only depth image .  

So i changed to   rgb_camera->getRGBImage(img_depth_)   but not work . 

How to change the observation to get the RGB images ? What is the lines i have to change ?

yun-long, 2021-03-29:
Hi,

1. **How to change the observation to get the RGB images? What are the lines I have to change?**

RGB image and depth image have different cv::Mat types. Specifically, RGB image use "cv::Mat rgb_imag_(frame_width_, frame_height_, CV_8UC3)" and depth image use "cv::Mat depth_image_(frame_width_, frame_height_, CV_32FC1)". 

2. @lorenzoferrini

lorenzoferrini, 2021-03-29:
Hi, if you change the type of the cv::Mat from CV_32FC1 to CV_8UC3 the values should already be in [0,255] so you won't need any rescaling:  images = images[0,:,:] should do the job

oggyfaker, 2021-03-29:
Thanks for your answer . As you told me : 

+ i changed ==> 
img_depth_ = cv::Mat::zeros(frame_width_, frame_height_, CV_8UC3);  in line 62 , and 151

+ i disable depth image with : 
rgb_camera->setPostProcesscing(
                    std::vector<bool>{false, false, false});
+ And:
//Update image observation
  rgb_camera->getRGBImage(img_depth_);
  // rgb_camera->getDepthMap(img_depth_);

But i have this error when i compile the env with python . I tried modify dimension of array by :
num_envs = 3 in (self.img_array.reshape((self.num_envs, self.frame_dim[0], self.frame_dim[1])))  
to make sure for 3 channels input but not work. 

RuntimeError: OpenCV(4.2.0) ../modules/core/src/matrix_wrap.cpp:1195: error: (-215:Assertion failed) !fixedType() || ((Mat*)obj)->type() == mtype in function 'create'

Hope to see your answer !