uzh-rpg/flightmare #108: dodge drone challenge: depth + state observation tips

jhurlbut, 2021-03-29:
I've tried installing a dictionary observation fork of stable-baselines but when I use that model in the DDC RL example code (even without using the depth images) the drone just keeps flipping over even after thousands of iterations of training. Are there any more tips on integrating both image and drone state observations into the code? 

I know this isn't the right place for this but if anyone out there wants to collaborate on the challenge, send me a message. Thanks!

hai-h-nguyen, 2021-03-30:
@jhurlbut how to contact you?

closed 2021-04-07 by jhurlbut

imgeorgiev, 2021-04-07:
I think it would be more helpful if the solution/advice on this issue are presented here publicly