uzh-rpg/flightmare #109: ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'

HRXWEB, 2021-03-30:
I was trying to install the dependencies for flight-mare, but when I typed the command:
pip install -r requirements.txt

it gave the error message:
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'

Seon-Kyuong-il, 2021-03-30:
I was encountered the same error. The file 'requirements.txt' didn't exist. 
So I just went over to the next step and there was no problem till now.

test-bai-cpu, 2021-03-30:
But in the instruction of installing flightmare: https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip , it does not say anything about the requirements.txt. It just list the required dependecies, which are tensorflow-gpu==1.14 and scikit-build.

HRXWEB, 2021-03-31:
> But in the instruction of installing flightmare: https://github.com/uzh-rpg/flightmare/wiki/Install-with-pip , it does not say anything about the requirements.txt. It just list the required dependecies, which are tensorflow-gpu==1.14 and scikit-build.

That's it! Thanks, the another instruction of installing flightmare: https://flightmare.readthedocs.io/en/latest/getting_started/quick_start.html#install-with-pip , it‘s a little different from Wiki

yun-long, 2021-03-31:
The requirements.txt has been removed. The documentation will be updated soon.

HRXWEB, 2021-03-31:
@yun-long Thanks! But I was encountered some error when I typed the command:
pip install tensorflow-gpu==1.14

it gave the error message:
ERROR: Could not find a version that satisfies the requirement tensorflow-gpu==1.14
ERROR: No matching distribution found for tensorflow-gpu==1.14

I use a Macbook Pro based on M1 chips, so I installed Archiconda3-0.2.3-Linux-aarch64.sh(https://github.com/Archiconda/build-tools/releases). Maybe that's the problem? How can I solve it?

khaledJabr, 2021-05-31:
@yun-long Any updates on this? thank you.