uzh-rpg/flightmare #112: docker: Error response from daemon: Unknown runtime specified nvidia.

Roopesh-Bharatwaj-K-R, 2021-04-01:
Hi I started with Ubuntu 20.04, After Running this in Terminal :
docker run -it -v path_to_dodge_drone_challenge_folder:/root/challenge/ --rm --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=all -p 10253:10253 -p 10254:10254 --name ddc-challenge uzhrpg/dodgedrone-challenge:base /bin/bash

docker: Error response from daemon: Unknown runtime specified nvidia.
See 'docker run --help'

got struck in this .

Praveendwivedi, 2021-04-02:
run without --runtime=nvidia

huiyulhy, 2021-04-04:
in terminal, check that nvidia-smi has an output (so your driver is working). After installing the docker, make sure that you have nvidia-docker installed as well (https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#docker). The latest version is nvidia-docker 2.0, so replace --runtime=nvidia with --gpus=all (or check out the other options for this flag)