uzh-rpg/flightmare #113: AttributeError: 'flightgym.QuadrotorEnv_v1' object has no attribute 'getFrameDim'

Praveendwivedi, 2021-04-02:
hey!
 I've followed all the steps as mentioned in the readme doc for ddc challenge. As mentioned in step 8 of the doc, when running
```
cd $FLIGHTMARE_PATH/flightrl/examples && python test_vec_env.py
```
I got following error:
```
[UnityBridge]  Initializing ZMQ connection!
[UnityBridge]  Initializing ZMQ connections done!
[VecEnv]       Flightmare Bridge is created.
Traceback (most recent call last):
  File "test_vec_env.py", line 94, in <module>
    main()
  File "test_vec_env.py", line 52, in main
    env = wrapper.FlightEnvVec(QuadrotorEnv_v1(dump(cfg, Dumper=RoundTripDumper), False))
  File "/root/miniconda3/envs/ENVNAME/lib/python3.6/site-packages/rpg_baselines/envs/vec_env_wrapper.py", line 13, in __init__
    self.frame_dim = self.wrapper.getFrameDim()
AttributeError: 'flightgym.QuadrotorEnv_v1' object has no attribute 'getFrameDim'
```

I checked about available methods of 'flightgym.QuadrotorEnv_v1' in python ide by running:
import flightgym
help(flightgym)

i got :
Help on module flightgym:

NAME
    flightgym

CLASSES
    pybind11_builtins.pybind11_object(builtins.object)
        QuadrotorEnv_v1
        TestEnv_v0
    
    class QuadrotorEnv_v1(pybind11_builtins.pybind11_object)
     |  Method resolution order:
     |      QuadrotorEnv_v1
     |      pybind11_builtins.pybind11_object
     |      builtins.object
     |  
     |  Methods defined here:
     |  
     |  __init__(...)
     |      __init__(*args, **kwargs)
     |      Overloaded function.
     |      
     |      1. __init__(self: flightgym.QuadrotorEnv_v1) -> None
     |      
     |      2. __init__(self: flightgym.QuadrotorEnv_v1, arg0: str) -> None
     |      
     |      3. __init__(self: flightgym.QuadrotorEnv_v1, arg0: str, arg1: bool) -> None
     |  
     |  __repr__(...)
     |      __repr__(self: flightgym.QuadrotorEnv_v1) -> str
     |  
     |  close(...)
     |      close(self: flightgym.QuadrotorEnv_v1) -> None
     |  
     |  connectUnity(...)
     |      connectUnity(self: flightgym.QuadrotorEnv_v1) -> bool
     |  
     |  curriculumUpdate(...)
     |      curriculumUpdate(self: flightgym.QuadrotorEnv_v1) -> None
     |  
     |  disconnectUnity(...)
     |      disconnectUnity(self: flightgym.QuadrotorEnv_v1) -> None
     |  
     |  getActDim(...)
     |      getActDim(self: flightgym.QuadrotorEnv_v1) -> int
     |  
     |  getExtraInfoNames(...)
     |      getExtraInfoNames(self: flightgym.QuadrotorEnv_v1) -> List[str]
     |  
     |  getNumOfEnvs(...)
     |      getNumOfEnvs(self: flightgym.QuadrotorEnv_v1) -> int
     |  
     |  getObsDim(...)
     |      getObsDim(self: flightgym.QuadrotorEnv_v1) -> int
     |  
     |  isTerminalState(...)
     |      isTerminalState(self: flightgym.QuadrotorEnv_v1, arg0: numpy.ndarray[bool[m, 1], flags.writeable]) -> None
     |  
     |  reset(...)
     |      reset(self: flightgym.QuadrotorEnv_v1, arg0: numpy.ndarray[numpy.float32[m, n], flags.writeable, flags.c_contiguous]) -> bool
     |  
     |  setSeed(...)
     |      setSeed(self: flightgym.QuadrotorEnv_v1, arg0: int) -> None
     |  
     |  step(...)
     |      step(self: flightgym.QuadrotorEnv_v1, arg0: numpy.ndarray[numpy.float32[m, n], flags.writeable, flags.c_contiguous], arg1: numpy.ndarray[numpy.float32[m, n], flags.writeable, flags.c_contiguous], arg2: numpy.ndarray[numpy.float32[m, 1], flags.writeable], arg3: numpy.ndarray[bool[m, 1], flags.writeable], arg4: numpy.ndarray[numpy.float32[m, n], flags.writeable, flags.c_contiguous]) -> bool
     |  
     |  testStep(...)
     |      testStep(self: flightgym.QuadrotorEnv_v1, arg0: numpy.ndarray[numpy.float32[m, n], flags.writeable, flags.c_contiguous], arg1: numpy.ndarray[numpy.float32[m, n], flags.writeable, flags.c_contiguous], arg2: numpy.ndarray[numpy.float32[m, 1], flags.writeable], arg3: numpy.ndarray[bool[m, 1], flags.writeable], arg4: numpy.ndarray[numpy.float32[m, n], flags.writeable, flags.c_contiguous]) -> None
     |  
     |  ----------------------------------------------------------------------
     |  Methods inherited from pybind11_builtins.pybind11_object:
     |  
     |  __new__(*args, **kwargs) from pybind11_builtins.pybind11_type
     |      Create and return a new object.  See help(type) for accurate signature.
    

FILE
    /root/miniconda3/envs/ENVNAME/lib/python3.6/site-packages/flightgym.cpython-36m-x86_64-linux-gnu.so

As you can see there is no method named 'getFrameDim'.
What should I do now :(?

yun-long, 2021-04-06:
Hi, this interface has been already implemented. Did you pull the latest version? 

`git pull`

yun-long, 2021-04-06:
Hi, this interface has been already implemented. 
see https://github.com/uzh-rpg/flightmare/blob/ddc_challenge/flightlib/src/wrapper/pybind_wrapper.cpp#L34

Make sure you have pulled the latest version

`git pull`