uzh-rpg/flightmare #114: ModuleNotFoundError: No module named 'ruamel'

Roopesh-Bharatwaj-K-R, 2021-04-03:
I Tried to  Install the flight mare flight lib following this link https://github.com/uzh-rpg/flightmare/wiki/Basic-Usage-with-Python#install-the-flightmare-library-flightlib 

then showed a error: Module Not Found
I use Microsoft Azure for Linux Os -Ubuntu 20.04, for entire process and installation, what is the next step from this, Thanks 
[image]

Praveendwivedi, 2021-04-04:
https://github.com/Azure/MachineLearningNotebooks/issues/1110 see if this helps....

yun-long, 2021-04-06:
`pip install ruamel.yaml` or 
`sudo pip install ruamel.yaml`

Roopesh-Bharatwaj-K-R, 2021-04-06:
@yun-long ,Thanks, I corrected that , after that it  showed - ModuleNotFoundError: No module named 'stable_baselines', for which i searched online and found [https://stable-baselines.readthedocs.io/en/master/guide/install.html](stable_baselines) and run this command: sudo apt-get update && sudo apt-get install cmake libopenmpi-dev python3-dev zlib1g-dev,

Below is the response for stable baselines issue:

(base) AzureUser@V1:~/anaconda3$ sudo apt-get update && sudo apt-get install cmake libopenmpi-dev python3-dev zlib1g-dev
Hit:1 http://azure.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://azure.archive.ubuntu.com/ubuntu focal-updates InRelease
Hit:3 http://azure.archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:4 http://security.ubuntu.com/ubuntu focal-security InRelease
Reading package lists... Done
E: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 3380 (unattended-upgr)
N: Be aware that removing the lock file is not a solution and may break your system.
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
(base) AzureUser@V1:~/anaconda3$

Now , Please help after this . Thanks.

Roopesh-Bharatwaj-K-R, 2021-04-14:
@yun-long , your suggestions for this.