uzh-rpg/flightmare #115: Segmentation Fault on running example code (ddc_challenge)

verma-rishabh, 2021-04-05:
Hi,
I have downloaded the docker image, and followed the instructions provided [here](https://github.com/uzh-rpg/flightmare/blob/ddc_challenge/README.md). The [example code](https://github.com/uzh-rpg/flightmare/blob/ddc_challenge/flightrl/examples/run_drone_control.py), runs fine with render=1, but gives Segmentation fault when render = 0.
This is the output of `./test_lib`
```
Running main() from /root/challenge/flightmare/flightlib/externals/googletest-src/googletest/src/gtest_main.cc
[==========] Running 22 tests from 6 test suites.
[----------] Global test environment set-up.
[----------] 4 tests from QuadrotorDynamics
[ RUN      ] QuadrotorDynamics.Constructor
Quadrotor Dynamics:
mass =             [1.2]
arm_l =            [0.25]
t_BM =             [ 0.177 -0.177 -0.177  0.177]
                   [-0.177 -0.177  0.177  0.177]
                   [0 0 0 0]
inertia =          [0.0281      0      0]
                   [     0 0.0281      0]
                   [     0      0 0.0437]
motor_omega_min =  [150]
motor_omega_max =  [2e+03]
motor_tau_inv =    [20]
thrust_map =       [1.33e-06  0.00384    -1.77]
kappa =            [0.016]
thrust_min =       [0]
thrust_max =       [11.2]
omega_max =        [6 6 6]

     1      1      1      1
 0.177 -0.177 -0.177  0.177
-0.177 -0.177  0.177  0.177
 0.016 -0.016  0.016 -0.016
[       OK ] QuadrotorDynamics.Constructor (0 ms)
[ RUN      ] QuadrotorDynamics.Dynamics
[       OK ] QuadrotorDynamics.Dynamics (0 ms)
[ RUN      ] QuadrotorDynamics.VectorReference
[       OK ] QuadrotorDynamics.VectorReference (0 ms)
[ RUN      ] QuadrotorDynamics.LoadParams
Quadrotor Dynamics:
mass =             [0.73]
arm_l =            [0.17]
t_BM =             [ 0.12 -0.12 -0.12  0.12]
                   [-0.12 -0.12  0.12  0.12]
                   [0 0 0 0]
inertia =          [0.00791       0       0]
                   [      0 0.00791       0]
                   [     0      0 0.0123]
motor_omega_min =  [150]
motor_omega_max =  [3e+03]
motor_tau_inv =    [1e+04]
thrust_map =       [1.33e-06  0.00384    -1.77]
kappa =            [0.016]
thrust_min =       [0]
thrust_max =       [11.2]
omega_max =        [6 6 6]

[       OK ] QuadrotorDynamics.LoadParams (0 ms)
[----------] 4 tests from QuadrotorDynamics (1 ms total)

[----------] 4 tests from Quadrotor
[ RUN      ] Quadrotor.Constructor
[       OK ] Quadrotor.Constructor (0 ms)
[ RUN      ] Quadrotor.ResetSimulator
[       OK ] Quadrotor.ResetSimulator (0 ms)
[ RUN      ] Quadrotor.RunQuadCmdFeedThrough
/root/challenge/flightmare/flightlib/tests/objects/quadrotor.cpp:98: Failure
Value of: quad.run(ctl_dt)
  Actual: false
Expected: true
/root/challenge/flightmare/flightlib/tests/objects/quadrotor.cpp:98: Failure
Value of: quad.run(ctl_dt)
[... 150 lines cut ...]
[ RUN      ] Logger.NoColorLogging
[Test]         This is a text stream log.
[Test]         Info:    This is an info log.
[Test]         Warning: This could be a warning, but just for demo.
[Test]         Error:   This could be an error, but just for demo.
[       OK ] Logger.NoColorLogging (0 ms)
[----------] 2 tests from Logger (0 ms total)

[----------] 3 tests from QuadState
[ RUN      ] QuadState.Constructor
[       OK ] QuadState.Constructor (0 ms)
[ RUN      ] QuadState.Accessors
[       OK ] QuadState.Accessors (0 ms)
[ RUN      ] QuadState.Compare
[       OK ] QuadState.Compare (0 ms)
[----------] 3 tests from QuadState (0 ms total)

[----------] Global test environment tear-down
[==========] 22 tests from 6 test suites ran. (5 ms total)
[  PASSED  ] 21 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] Quadrotor.RunQuadCmdFeedThrough

 1 FAILED TEST
```

jhurlbut, 2021-04-06:
I was also getting a seg fault with render set to zero and solved it by commenting out the code in set_objects_densities inside vec_env_wrapper.py. 

It seems like there is a bug where self.render is not set to false so the code tries to send a message to the standalone app even though it's not running. 

Not sure if it's the same issue for you.

verma-rishabh, 2021-04-06:
> I was also getting a seg fault with render set to zero and solved it by commenting out the code in set_objects_densities inside vec_env_wrapper.py.
>
> It seems like there is a bug where self.render is [...]

Yeah that was the issue, thanks.

closed 2021-04-06 by verma-rishabh