uzh-rpg/flightmare #117: Flight_pilot + Camera

sbhatti915, 2021-04-07:
I am having trouble integrating the flight_pilot.cpp and camera.cpp to have Flightmare move with our controller and to see the images produced from the camera. Has anyone been able to successfully integrate these 2?

JosephFeld, 2021-07-13:
I got this working! Right now it's in my private repo - if you still want this dm me.

hanif-hazemi, 2021-12-30:
Hi! @JosephFeld. Would you mind sharing how you do this? I copy the nodes (camera and rviz) from the camera.launch file to rotors_gazebo.launch but no luck so far (flight_pilot_node crashes). Appreciate your help.. @yun-long

sirengarcia, 2022-08-18:
> I got this working! Right now it's in my private repo - if you still want this dm me.

Sorry to bother u @JosephFeld, I've been struggling with the implementation of this feature for a month now, would you mind sharing your method please?

8ajpai, 2022-09-12:
Hey @sirengarcia @hanif-hazemi did you figure it out? @JosephFeld How do i dm you?

sirengarcia, 2022-09-20:
> Hey @sirengarcia @hanif-hazemi did you figure it out? @JosephFeld How do i dm you?

yes, I have solved it. I used the rl modules. Do u need help still?

sirengarcia, 2022-09-20:
yes, I have solved it. I used the rl modules. Do u need help still? @8ajpai

8ajpai, 2022-09-23:
Hey @sirengarcia that would be helpful. Can we talk over email? s7utbajp@uni-bonn.de